iNavFlight/inav #5166: OMNIBUSF4PRO_LEDSTRIPM5 Barometer not working INAV 2.3.0 RC1

breadoven, 2019-10-27:
## Current Behavior
FC board was previously working fine using version 2.2.1.

After flashing version 2.3.0 RC1 for OMNIBUSF4PRO_LEDSTRIPM5 barometer BMP280 showing Red in Configurator and not responding in sensor tab. Setup Arming status showing hardware failure.  Tried changing barometer setting in Configuration, Auto, Fake BMP280, nothing works.

Baro works fine after reflashing back to version 2.2.1,

## Steps to Reproduce
Flash version 2.3.0 RC1

## Expected behavior

## Suggested solution(s)
Turn the Baro Blue

## Additional context

---

- FC Board name and vendor: Omnibus F4 Pro (Fake)
- INAV version string: Target: OMNIBUSF4PRO LEDSTRIPM5
Name/Version: 
Binary: inav_2.3.0_OMNIBUSF4PRO_LEDSTRIPM5.hex
Date: 2019-10-27 15:6
State: release-candidate

Jetrell, 2019-10-28:
Just to confirm. I'm having the same issue with an Airbot OmnibusF4Pro (bmp280) as well.

doughadfield, 2019-10-28:
Hi, I have the same issue with all three of my Omnibus F4 Pro boards (two v3 and one v2). Works fine on iNAV V2.2.1. Here's the status output for one of the boards WORKING on 2.2.1:
[OmnibusV4Pro221.txt](https://github.com/iNavFlight/inav/files/3779261/OmnibusV4Pro221.txt)
...and here's the status for the same board when flashed with v2.3.0(rc) (after selecting BMP280 as barometer type):
[OmnibusV4Pro330rc.txt](https://github.com/iNavFlight/inav/files/3779276/OmnibusV4Pro330rc.txt)

breadoven, 2019-10-28:
Had a look at the changes in 2.3.0 and thought #4977 "Fix baro and SPI flash on OMBF4" might have caused the problem. Managed to rebuild the hex for my board from the 2.3.0 source files with the #4977 changes in the Target.h file removed and low and behold the Baro is now working ... which is a miracle given I didn't entirely understand the instructions given on here about building your own firmware. No smoke from the board though so far which has got to be a good sign.

doughadfield, 2019-10-29:
Nicely done,  Breadoven!! - so you've successfully isolated the issue to a single specific change - excellent! I'm wondering what purpose the #4977 fix served, and what you're now missing with your custom build. Hopefully the developers can resolve the #4977 issue before final release.

putimir, 2019-10-29:
Same here, original Airbot Omnibus F4V5, OMNIBUSF4V3 target...looking forward to fix

digitalentity, 2019-10-31:
Oh my. The million variants of OMNIBUS F4 boards give me the shivers :man_facepalming:

putimir, 2019-11-01:
Haha, no need to hurry, I bet we all have twice or three times too many models to fly...

lucky13hc, 2019-11-02:
> Had a look at the changes in 2.3.0 and thought #4977 "Fix baro and SPI flash on OMBF4" might have caused the problem. Managed to rebuild the hex for my board from the 2.3.0 source files with the [...]

could you share the file or we must wait for rc2?

tiriad, 2019-11-03:
For those who want to try this new 2.3 RC I have applied the changes to detect the barometer again in our baro SPI no I2C and corrected the ledstrip in the new versions with softserial in M5 & M6. Working on a Omnibus F4v5

[inav_2.3.0_OMNIBUSF4V3_S5_S6_2SS.zip](https://github.com/iNavFlight/inav/files/3801836/inav_2.3.0_OMNIBUSF4V3_S5_S6_2SS.zip)
[inav_2.3.0_OMNIBUSF4PRO_LEDSTRIPM5.zip](https://github.com/iNavFlight/inav/files/3801837/inav_2.3.0_OMNIBUSF4PRO_LEDSTRIPM5.zip)

putimir, 2019-11-04:
Works for me...(Airbot Omnibus F4v5, OMNIBUSF4V3_S5_S6_2SS)

breadoven, 2019-11-08:
Just to confirm inav_2.3.0_OMNIBUSF4PRO_LEDSTRIPM5.zip above from Tiriad doesn't work on my board even though that is the version I've always used. All seemed fine until I actually tried to fly and realised the servos weren't working (never noticed when testing in the house). PWM Init Error on the OSD and something about a lack of PWM outputs/timers in CLi status when I checked later. Reflashed my Baro hacked version of 2.3.0 and it's working again. Must be to do with the Ledstrip M6 and M6 changes. My board is an Omnibus F4 V3 Pro clone but without the current measuring resistor attached, is this a Flip 32 ? Hard to know exactly what you got. Never bothered with the LEDs so don't know if the pinouts work or not or if they're supposed to.

tiriad, 2019-11-08:
Could you explain a little more how is your configuration?, What do you have connected in each output PWM?, specifically in the PWM5. OMNIBUSF4PRO_LEDSTRIPM5 target is to have a led in the PWM5, is the only serious modification that takes, I think that if you have something that is not a led in the PWM5 the recommended version is OMNIBUSF4PRO. Please try this compiling and comment, otherwise it may be necessary to reverse some change.
[inav_2.3.0.test_OMNIBUSF4PRO.zip](https://github.com/iNavFlight/inav/files/3825912/inav_2.3.0.test_OMNIBUSF4PRO.zip)

breadoven, 2019-11-08:
My board looks like the one shown at the start of this F4.md.
https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md

I used OMNIBUSF4PRO_LEDSTRIPM5 because it's an eBay clone. I've used PWM1 for a motor, PWM2 unused and PWM3 to PWM6 are connected to servos. This all seems to work. Didn't really understand the LED bit and don't have any LEDs connected to the dedicated LED pin so don't know what it does. Is this saying PWM5 on clone boards won't work as a PWM output, just works for LED ? In which case where is the PWM5 output, or is it missing ? I thought the Softserial went on C5 and C6 rather than PWM5 and PWM6 ? However, C5 and C6 pads aren't on my board, don't exist, they only seem to be on the boards with the current resistor mounted.

I haven't tried OMNIBUSF4PRO so don't know if that would work as well. Guess I could give it a try. Too many versions of these boards, all called the same thing.

tiriad, 2019-11-08:
If it is the board in the wiki photo without current meter resistor is the TARGET OMNIBUSF4V3 there are many clones. This firmware is used a lot in multicopters and as you say the target F4PRO_LEDSTRIPM5 uses is for the clone of the F4 PRO (with current meter) that does not work the dedicated pin to led and is placed in the servo out 5. So if you use all outputs for servos the target OMNIBUSF4V3 without SoftSerial or LEDS is for you
[inav_2.3.0.test_OMNIBUSF4V3.zip](https://github.com/iNavFlight/inav/files/3826499/inav_2.3.0.test_OMNIBUSF4V3.zip)

breadoven, 2019-11-08:
LOL ... I only assumed it was the Pro because it said that in the eBay listing and I thought the Pro had the Baro with the current meter resistor only fitted depending on the clone type. Confusing stuff.

Anyway I'll try OMNIBUSF4V3  and see if it still works correctly.

breadoven, 2019-11-09:
OK the inav_2.3.0.test_OMNIBUSF4V3.zip firmware works fine, no problems. So thanks for that and clarifying the board differences, much appreciated.

digitalentity, 2019-11-23:
Closing as obsolete, see #5187

closed 2019-11-23 by digitalentity

iz3man, 2020-11-01:
May I reopen this? After about 1-2 years I resurrected my 5" quad with the omnibus f4 pro board and flashed 2.5.2 (ledstripm5 fw version) to it. All works fine, but the baro is NOT recognized a remains RED. 
Any idea what I can do other then downgrade or order a new FC?
Thanks