iNavFlight/inav #5171: RX & OSD support for RSSI DBM and RX Link Quality

pitts-mo, 2019-10-28:
Some RX are able to provide more information about the control signal just a simple RSSI percentage estimate. 

Making detailed control signal status available in the OSD would be helpful.

Suggest reference betaflight code to integrate support for the below OSD elements into inav along with the support needed to get this data from supporting rx.
OSD_RSSI_DBM_VALUE
OSD_LINK_QUALITY

This feature would provide pilot greater awareness of true control signal reliability and should reduce instances of control loss.

pitts-mo, 2019-10-28:
The primary receiver type I had in mind is CRSF. Implementing warnings and stats for these values would be helpful as well. -p

SRWieZ, 2019-10-28:
I think it's done here : #5098

pitts-mo, 2019-10-28:
I swear I searched before creating this issue :-). Yes, #5098 appears to cover this issue. Thanks, -p

closed 2019-10-28 by pitts-mo