iNavFlight/inav #5175: DSHOT/ESC Telemetry + Airplane mixer

dupedup, 2019-10-29:
I'm using FC Omnibus F4V6 / OMNIBUSF4V6+ new INAV 2.3.0 RC1 + ESC DYS Aria 35 / BL_Heli32 ver 32_64. I have wired TE output on ESC to RX4 on Omnibus FC.
 All is working perfectly if the mixer is set as "Multirotor" and e.g. Quad X in INAV -  I can set a DSHOT protocol (eg DSHOT600) and use  ESC Telemetry (eg RPM in my OSD). 
But if I set up the mixer as "Airplane" I'm not able to set any DSHOT protocol in the configuration n INAV. It may be  strange using DSHOT with an airplane with fixed wing but I use it.
I try it with FIREWORKSV2 with the same result. In Betaflight I can use DSHOT and ESC Telemetry with Custom Airplane mix setting normally.
Is any reason to disable DSHOT with Airaplane mixer in INAV  or I do anything wrong?

DzikuVx, 2019-10-30:
PLease configure only as many motors in Mixer as you are using. Default airplane mixer has 2 motors and if you use only one....

dupedup, 2019-10-31:
Thank you for your reply, I have configured only one motor, the second  I deleted  from default AIRPLANE mixer conf. However if I change the ESC protocol to DSHOT and reboot, after reboot is in  the configuration doggedly STANDARD/490 Hz again:( May be a problem of OMNIBUS FC, some controllers have reputedly only one group PWM outputs with only one output protocol. But in the  mix AIRPLANE I would like to have two groups - motor/motors as DSHOT and servos as the standard PWM. I will try MATEK 405SE.

dupedup, 2019-11-02:
I have tried MATEKF405SE, I can change ESC protocol to DSHOT (DSHOT remains in the configuration after reboot and mixer AIRPLANE now ). However the motor is not initialized. Unfortunately the motor is not initialized neither with AIRPLANE nor MULTIROTOR with DSHOT now… If I change the ESC  protocol to STANDARD or any analog (ONESHOT) - no problems.

Ozyris, 2020-04-18:
Same issue. Airplane with Airbot Omnibus F4 nano V6, unable to set DShot protocol.

Ozyris, 2020-04-19:
Seems like dshot not supported on this board. https://github.com/iNavFlight/inav/blob/master/src/main/target/FIREWORKSV2/target.c

closed 2020-12-11 by DzikuVx