iNavFlight/inav #5176: Compass can't be calibrated

GGYHCLZ, 2019-10-30:
My  Board  is   Matek F405.

My Compass can't be calibrated.
But  if  i  use   BETA  Firmware.
The  Compass   can   worked  very   well.

GGYHCLZ, 2019-10-30:
[image]

GGYHCLZ, 2019-10-30:
[image]

GGYHCLZ, 2019-10-30:
After Save  And  Rebort

[image]

GGYHCLZ, 2019-10-30:
[image]

GGYHCLZ, 2019-10-30:
After  calibrated
[image]

GGYHCLZ, 2019-10-30:
[image]

GGYHCLZ, 2019-10-30:
[image]
[image]

GGYHCLZ, 2019-10-30:
[image]
[image]

GGYHCLZ, 2019-10-30:
[image]
[image]

GGYHCLZ, 2019-10-30:
[image]
[image]

GGYHCLZ, 2019-10-31:
I use it

[image]

DzikuVx, 2020-06-11:
this is fixed in 2.5

closed 2020-06-11 by DzikuVx