iNavFlight/inav #5182: inav build error

nurettinaci, 2019-10-31:
I tried everything, but I couldn't build it.

error  : 

`$ make TARGET=SPRACINGF3
C:/Ruby26-x64/lib/ruby/2.6.0/open3.rb:213:in `spawn': No such file or directory - C:\dev\7 2018-q2-update\bin/arm-none-eabi-g++.exe -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Wdouble-promotion -flto -fuse-linker-plugin -Os -I./src/main -I./src/main/target -I./lib/main/STM32F3/Drivers/STM32F30x_StdPeriph_Driver/inc -I./lib/main/CMSIS/Core/Include -I./lib/main/STM32F3/Drivers/CMSIS/Device/ST/STM32F30x -I./lib/main/MAVLink -I./src/main/target/SPRACINGF3 -ggdb3 -DDEBUG -std=c++11 -Wall -Wextra -Wunsafe-loop-optimizations -Wdouble-promotion -Werror=switch -ffunction-sections -fdata-sections -fno-common -DSTM32F303xC -DSTM32F303 -DARM_MATH_MATRIX_CHECK -DARM_MATH_ROUNDING -D__FPU_PRESENT=1 -DUNALIGNED_SUPPORT_DISABLE -DARM_MATH_CM4 -DFLASH_SIZE=256 -DHSE_VALUE=8000000 -DUSE_STDPERIPH_DRIVER -DSTM32F3 -DSPRACINGF3 -D__FORKNAME__=inav -D__TARGET__=SPRACINGF3 -D__REVISION__=af08641f9 -I./obj/main/SPRACINGF3 -dM -E -o ./obj/main/SPRACINGF3/stamp ./obj/main/SPRACINGF3/stamp.cpp (Errno::ENOENT)
        from C:/Ruby26-x64/lib/ruby/2.6.0/open3.rb:213:in `popen_run'
        from C:/Ruby26-x64/lib/ruby/2.6.0/open3.rb:101:in `popen3'
        from C:/Ruby26-x64/lib/ruby/2.6.0/open3.rb:281:in `capture3'
        from C:/dev/inav/src/utils/compiler.rb:91:in `run'
        from ./src/utils/build_stamp.rb:66:in `hash_config'
        from ./src/utils/build_stamp.rb:44:in `stamp'
        from ./src/utils/build_stamp.rb:89:in `<main>'
make: *** [Makefile:316: obj/main/SPRACINGF3/build.stamp] Error 1
`

breadoven, 2019-11-01:
Have you added the location of "arm-none-eabi-g++.exe" to the Windows Path environment variable ?

nurettinaci, 2019-11-01:
@breadoven yes i added

breadoven, 2019-11-01:
If it has "%PATH%;" at the start, as it mentions in the Windows build.md on here, then delete it. Just use the path starting with C:\. When I used %PATH%; it didn't work, couldn't find the file.

nurettinaci, 2019-11-02:
I checked.
https://imgyukle.com/i/Eew9X6
https://imgyukle.com/i/Eewfep

DzikuVx, 2019-11-02:
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20with%20Linux%20Subsystem.md

This is proven to be working. Other Windows based build methods are either not working or pain in the ass. We will cleanup dev. docs

breadoven, 2019-11-02:
I used the method given in:
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md

But found the following problems:
The  GNU ARM Toolchain link is wrong. I searched for  GNU ARM Toolchain instead and downloaded the install package from the search link.
The PATH Windows environment variable: **%PATH%;C:\dev\gcc-arm-none-eabi-*version number*\bin** doesn't work with the **%PATH%;** bit included, works fine without.

I just downloaded the Source file for version 2.3.0 RC1 and used that to build from. Gives a load of Fatal Errors to do with it not being a Github repository etc, etc but the hex it produced still works all the same. Checking out from Github seems to just give you Version 2.2.1. source files ... the last stable version ?

tiriad, 2019-11-07:
I have that same mode working
Latest compiler version
[Official Web](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads)
[Link to the ZIP](https://developer.arm.com/-/media/Files/downloads/gnu-rm/8-2019q3/RC1.1/gcc-arm-none-eabi-8-2019-q3-update-win32.zip?revision=2f0fd855-d015-423c-9c76-c953ae7e730b?product=GNU%20Arm%20Embedded%20Toolchain,ZIP,,Windows,8-2019-q3-update)
In Cygwin window  change to the directory you want, and inside the folder run the next
git clone --single-branch --branch development https://github.com/iNavFlight/inav.git  
with the latest corrections and changes or
git clone --single-branch --branch 2.3.0 https://github.com/iNavFlight/inav.git
in make/tools.mk edit and change your gcc version in this case 8.3.1, if not it give you a version error
and finally make TARGET=YOURTARGET

DzikuVx, 2019-11-07:
Please do not use Cygwin. Use WSL or Docker instead

closed 2019-11-12 by nurettinaci