iNavFlight/inav #5183: Implement CRSF to TBS VTxs

swissfreek, 2019-11-01:
TBS' new 32-bit VTxs support CRSF direct to the VTx, which enables new features (such as MyVTx) and supposedly more features in the future. But right now it requires connecting the VTx directly to the Rx, which eliminates the ability to (for instance) use global variables to adjust transmit power in flight as you get farther from home, or to see your channel and power level in the OSD.

Is there any possibility of incorporating CRSF communication to the VTx as well as to the Rx? I could be mistaken, but it's my understanding that CRSF is like a network, in that anything connected can see anything else, so in theory if the Rx spoke to the FC via CRSF and the FC spoke to the VTx via CRSF, the Rx (and hence the radio module) could "see" the VTx, thereby allowing myVTX and whatever else comes along, while still maintaining the connection to the FC.

wx4cb, 2019-11-01:
don't you just tee them off on the same line. as a bus ? From page 9 of the Unify EVO manual

[image]

swissfreek, 2019-11-01:
There’s no flight controller in that image though. The top row are all Rxs, bottom row are all VTxs. So it’s 1:1.

wx4cb, 2019-11-01:
> There’s no flight controller in that image though. The top row are all Rxs, bottom row are all VTxs. So it’s 1:1.

it shouldn't matter if there is an FC in there or not, they're all tied in on a bus configuration.

swissfreek, 2019-11-01:
That’s true, but it still doesn’t solve the issue that the FC does not know how to send the VTx commands over CRSF.

swissfreek, 2019-11-04:
OK, based on comments from Trappy in a similar issue thread on the BetaFlight GitHub, I see that the CRSF protocol isn’t open at this time, but that it will be opened up once the features have stabilized, and there’s currently no timeline for that to happen. So it sounds like there’s nothing to be done yet, but sometime in the future it should be possible. So I can wait patiently until things are ready.

closed 2019-11-04 by swissfreek