iNavFlight/inav #5189: MAMBA F722S  Flight Controller

caasisaac, 2019-11-05:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

digitalentity, 2019-11-06:
In progress

namirda, 2019-11-06:
I have written an INav target for the MambaF722  which works well for me. I cannot guarantee that it handles all configurations but for my specific requirements it works fine. 

If my code would be helpful then please tell me how to contribute it - I am not familiar with github pull requests.

Thanks

nandibrenna, 2019-11-28:
@namirda can you please share your code?
Thanks

namirda, 2019-11-28:
OK - here is a zip file containing the files for my MAMBAF722 target.

I am using it with GPS, compass (HMC5883), barometer (MS5611), Bluetooth and I2C without problems. I had trouble with dshot and so am using multishot. I have not tried soft-serial
 
[MAMBAF722.zip](https://github.com/iNavFlight/inav/files/3902972/MAMBAF722.zip)

digitalentity, 2019-11-29:
@namirda I've added some of your fixes into my variant of the target (I forgot about target-config, shame on me). Please take a look #5231. Should be flyable.

namirda, 2019-11-30:
@digitalentity  All looks fine. I had omitted the RX6 def in config.c!! 

A couple of questions while I have the attention of the expert!:

1) As I mentioned in an earlier post, dshot does not work for me with this board. Motor 4 does not spin.  Do you have any idea why that might be?

2) This board does not have a solder pad for TX1 - it only has  SBUS and PPM. I am therefore forced to move FShot to Uart2 instead. This works OK but is a complete waste of a Uart.
How would I modify the code so that I can use the PPM pad as TX1? Is that possible?

Thanks

digitalentity, 2019-11-30:
1. I'm looking into DSHOT issue currently.
2. That's possible, AFAIR F722 has the possibility to swap TX/RX pins on the UART, but we don't have underlying code to allow that. Worth implementing :thinking:

digitalentity, 2019-11-30:
DSHOT seems to work now, @namirda can you double-check?

namirda, 2019-11-30:
Excellent - seems OK now! 

All four motors now turn on the bench using all DSHOT300/600/1200 - not flown yet. 

I see you have made a correction to the dmavar parameter in the TIM_DEF call for S3. Was that the cause of my DSHOT problem? 

What would have been the effect of that error when flying Multishot? I have been having some instabilities in flight. 

Will fly your new MBF7 target later today if the weather holds up.

Thanks!

digitalentity, 2019-12-01:
> I see you have made a correction to the dmavar parameter in the TIM_DEF call for S3. Was that the cause of my DSHOT problem?

Yes. M4 and M3 basically shared the same DMA which is impossible.

> What would have been the effect of that error when flying Multishot? 

Unlikely. Multishot and other PWM-based protocols don't use DMA.

closed 2019-12-05 by digitalentity

dimkasta, 2019-12-12:
Guys is this board available on the configurator somehow? I installed the latest version yesterday, but I could only see the F405 board in the dropdown list

bizio996, 2020-01-30:
The Diatone support replied:

I2C Question:

Very easy to change USART3 ports to I2C using a couple commands in the CLI.   I have attached a file with the commands.   Just need to copy and paste them into the CLI, press enter, then type SAVE and enter.  This will change to the following:

USART3 TX = I2C SCL
USART3 RX = I2C SDA

SCL is the clock signal for the I2C
SDA is the data signal.  It is bi-directional and sends the data bit by bit.

Here are the commands also:

resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
set dashboard_i2c_bus = 2
set mag_hardware = AUTO
set baro_hardware = AUTO

namirda, 2020-01-31:
@bizio996  I don't think any of this is required. The new MAMBAF722 target in iNav 2.4 handles all of that stuff

I have been flying with the MAMBAF722 target that is now available as 2.4 RC-2 for several months with no problems.  No additional CLI fiddling is required.  

Simply connect your I2C devices in parallel to UART3 and it should work.

bizio996, 2020-01-31:
how did you activate the compass without remapping the pads?  this fc does
not have the sca and scl pads

bizio996, 2020-01-31:
I soldered the scl cable on the tx3 pad and the sda ​​cable on the rx3 pad, but the magnetometer does not activate on the inav configurator

namirda, 2020-01-31:
[image]

Yes it does have scl/sda pads - not very convenient for sure but they do exist. However, you can simply use the TX3/RX3

namirda, 2020-01-31:
I am using a custom build of iNav 2.3 using the same MAMBAF7 target which is now available in 2.4 RC-2. Everything is working fine. I have not tried using 2.4 RC-2 yet. 

I am using a BN-880 compass/GPS and an MS5611 barometer both connected to UART3 - all works well

bizio996, 2020-02-03:
I too would like to use mamba f722s with inav, but the protocols esc 300 and 600 works really badly, now it work with multishot. Please you see the video on test dshot 300 and dshot 600 don't work, when I get up throttle one motor is stopped is this happened in the fly! 
i also tried with betaflight and it all works with my esc dshot 300 and 600

[https://youtu.be/9NiGlqb5OEA](url)

bizio996, 2020-02-03:
here you can see some flight tests with multishot protocol

https://youtu.be/NI2bVEjLd-A    (1° test multishot inav2.4 pid stock)
https://youtu.be/v0TwZOdUE_4  (2° test multishot inav2.4)
https://youtu.be/iA1zj6HkVEk    (3° test multishot inav2.4)
https://youtu.be/RuWK8nTHk3Y (4° test multishot inav2.4)
https://youtu.be/NOmo9JLOlEw  (5° test multishot inav2.4)

DarkSoulRL, 2020-03-05:
> [image]
> 
> Yes it does have scl/sda pads - but they are tiny and not easy to see or use. However, you can simply use the TX3/RX3 pads - the mapping is all handled by the target code.

These little pads do not work. The pins of the pa13 SCL pa14 SDI processor are connected to them, and in Inav these resources are free. 
 compass and an MS5611 barometer both connected to UART3 - all works well, 
but I want to free UART3

eben80, 2020-04-29:
> @bizio996 I don't think any of this is required. The new MAMBAF722 target in iNav 2.4 handles all of that stuff
>
> I have been flying with the MAMBAF722 target that is now available as 2.4 RC-2 for [...]

Hi,
Would this be the same for the MAMBAF722 Mini?

The mapping is:
USART3 TX = I2C SCL
USART3 RX = I2C SDA 

?
Thanks

namirda, 2020-04-29:
I don't have the mini, but as far as I can see from the diatone wiring drawings there is no solder pad for Rx3?? If that is the case you have a problem.

eben80, 2020-04-30:
Yes I saw that too. Would it be possible to remap any of the other UARTS
that have RX and TX in code?
I've successfully compiled the target myself but I'm not sure if there are
hardware limitations in doing this.
If it's possible then maybe the mini could be a separate target.

namirda, 2020-04-30:
I'm really no expert at this pin mapping stuff but I don't think that works.

If you look at the specs for the STM32F722 (https://www.st.com/resource/en/datasheet/stm32f722ic.pdf) I think you find that PB10 and PB11 are the only possibilities for I2C - so you need to get at those pins if you want I2C to work.

The pad marked VTX on the Mamba F722 Mini Mark 2 is the video input to the VTX - not a uart at all.

eben80, 2020-05-01:
> I'm really no expert at this pin mapping stuff but I don't think that works.
> 
> If you look at the specs for the STM32F722 (https://www.st.com/resource/en/datasheet/stm32f722ic.pdf) I think you find that PB10 and PB11 are the only possibilities for I2C because I presume Uart1 is in use by your receiver - so you need to get at those pins if you want I2C to work.
> 
> The pad marked VTX on the Mamba F722 Mini Mark 2 is the video input to the VTX - not a uart at all.

Thank you for your advice.
I went ahead and found the PB11 pin on the STM32 and soldered a wire onto it to expose RX3.
[image]

The Compass didn't want to show up so I soldered some 1K pull-up resistors to the SDA and SCL lines to the 3.3V from the built-in voltage regulator.
[image]
The Compass then showed up but was intermittent as was shown below.
[image]

I then lowered the i2c speed to 200kHz and then the connection was stable and without dropouts. I reckon if I pay around with the value of the resistors I will be able to get it working at 400kHz if needed.

So in short, it IS possible to use I2C on the MAMBA F722 Mini, but not without some hardware modification.

bizio996, 2020-05-01:
Hi @DarkSoulRL I read that you used an external barometer and it worked for you, could you tell me on which pad you welded the external barometer MS5611? Thanks so much for your help

MaxMihawk, 2020-06-28:
This is a very confusing thread because you talk about 3 different FC´s. I have a F722S (https://github.com/iNavFlight/inav/wiki/Welcome-to-INAV,-useful-links-and-products --> that is the one you linked from the INAV supported page as "F722") which is NOT a F722. The F722 comes with a F506 ESC, the F722S comes with a F50Pro ESC. The Padlayout is not exactly the same either. And then their is a F722 mini of course. This thread seems to be for the F722S.

I used the Mamba target F722 and my resources looking like this:

**`IO:
----------------------
A00: SERIAL4 UART TX
A01: SERIAL4 UART RX
A02: FREE 
A03: FREE 
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR3 OUT
A09: MOTOR4 OUT
A10: FREE 
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FLASH CS
B00: FREE 
B01: FREE 
B02: BEEPER OUT
B03: LED STRIP OUT
B04: FREE 
B05: SPI3 MOSI
B06: SERIAL1 UART TX
B07: SERIAL1 UART RX
B08: FREE 
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: FREE 
C04: MPU EXTI
C05: FREE 
C06: FREE 
C07: SERIAL6 UART RX
C08: MOTOR1 OUT
C09: MOTOR2 OUT
C10: SPI3 SCK
C11: SPI3 MISO
C12: FREE 
C13: FREE 
C14: LED2 OUT
C15: LED1 OUT
D02: FREE `**

This looks very empty.... whats wrong? If this resources would be set properly i would remap the I2C SCL / SDA to a free UART or propably just use B10 & B11 if I could identify what´s their position on the FC (so what solderpads are B10 & B11). Any Tips? And why is my resource drop so empty?!?

DarkSoulRL, 2020-06-28:
Mamba f722: 
do not connect anything else to UART 3. SDA SCL only
B10 - SERIAL_TX 3 - SCL
B11 - SERIAL_RX 3 - SDA

Mitch-bit, 2020-07-02:
Hi there 🙋‍♂️ I hope somebody is still on this .. 
at first .. I tried to understand what you all posted , but to be honest .... that was way to much for someone who just started FPV a few months ago.
But I do have the same problem:
F722s and I would like to ad a Matek GPS with compass so that I could finish my first own build .
Now ... for a beginner .. what do I have to do ..
I’m really sorry but in Germany FPV is not that popular yet ... at least in my region ..
Please help me

MaxMihawk, 2020-07-03:
It´s very popular in germany. GPS works out of the box. For GPS you need to add the GPS wires to an UART. If you want to use the build-in magnetometer too (I think Matek has one), you need to add it to the I2C bus, which means SDA & SCL and in the F722S case UART3. (See DarkSoulRL post). We had this discussion in german too.... --> https://fpv-racing-forum.de/thread/14010-mamba-f722s-i2c-mapping/

MaxMihawk, 2020-07-07:
I´m super confused now. I want to set softserial on PPM (pin B09) for S.Port telemetry on the F722S. But I don´t get it to work and i read, that remapping does not work with INAV?! I have a F405 mk2 with exact the same configuration working with INAV. Did i miss something? What is the correct CLI command, for this change?