iNavFlight/inav #5191: Waypoints won't save to eeprom V 2.3.0 RC1

breadoven, 2019-11-07:
## Current Behavior
Waypoints created in Mission Control save to FC board but won't save to eeprom.

## Steps to Reproduce
1. With USB power, create waypoints in Mission Control and save to FC (or enter directly through CLi input)
2. In Mission Control save to eeprom, eeprom saved message shown. Check waypoints are saved to FC board by entering "wp" in CLi, should appear in CLi correctly. If entered directly through CLi save using save command.
3. After board has rebooted from CLi try and load waypoints from eeprom in Mission Control. Nothing loads.
4. Try and load waypoints to FC then check using CLi "wp". CLi waypoint list is empty.
5. With board powered from flight battery Repeat 1. and save using stick command, Board beeps indicating saved but waypoints not saved after board rebooted, CLi waypoint list is empty after trying to load from eeprom to FC in Mission Control..

## Expected behavior
Waypoints should save to eeprom as version 2.2.1.

## Suggested solution(s)

## Additional context
After reflashing iNav 2.2.1 waypoints load and save to eeprom correctly so not a board problem.

---
INAV/OMNIBUSF4PRO_LEDSTRIPM5 2.3.0 Oct 28 2019 / 20:28:58 ()
GCC-8.3.1 20190703 (release) [gcc-8-branch revision 273027]

- FC Board name and vendor: Omnibus F4 V3 Pro (Clone)
- INAV version string: OMNIBUSF4PRO_LEDSTRIPM5 2.3.0

breadoven, 2019-11-07:
Forgot to check what happens when you load the WP Mission using stick commands. When you do this the waypoints load correctly so they are being saved to eeprom. It's just Configurator that isn't loading them from eeprom correctly so it's a Configurator issue. Still least it works just don;t trust what Configurator is telling you when you try and load waypoints from the board.

stronnag, 2019-11-07:
> Forgot to check what happens when you load the WP Mission using stick commands. When you do this the waypoints load correctly so they are being saved to eeprom. It's just Configurator that isn't [...]

You should close this issue and raise a Configurator issue. It's not a firmware bug; mwp and impload save and restore missions to/from eeprom without issue.

closed 2019-11-07 by breadoven