iNavFlight/inav #5192: F411-WING iNav 2.2.1 can't arm with good gps fix and HDOP

IVData, 2019-11-07:
## Current Behavior
I have a plane with a F411-Wing with known issues with GPS interference from runcam split style cameras. However the signal is usually good enough to get 7-10 satellites and a HDOP of around 2.

Recently I've been unable to arm (osd says waiting for GPS fix), even though I have a fix (checked in GPS tab) and HDOP is ~1.9m. Are there some other GPS parameters that are checked before arming is allowed?

## Steps to Reproduce

1. Power up.
2. Wait for satellites.
3. Still can't arm even with fix.

## Expected behavior

Should be able to arm with a 3D fix and low HDOP.

## Additional context
This plane has always had trouble *getting* a fix, but once it did I have always flown without issue. It seems like this new problem of not being able to arm came along with 2.2.1 but I'm not sure.

Diff all: https://pastebin.com/huRfWmjt

---

```
# version
# INAV/MATEKF411 2.2.1 Jul  3 2019 / 22:50:30 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]
```

IVData, 2019-11-08:
Some work has been done on shielding the cameras.
Here is another DVR clip out in the field: https://www.youtube.com/watch?v=erhPsvsmLgg
14 satellites and cant arm???

Jetrell, 2019-11-10:
Have you looked under the GPS tab in the Configurator > `GPS Status` to see if there are any errors, timeouts or data message issues.
The GPS TX line maybe okay, while the RX line might be intermittent.

IVData, 2019-11-19:
Closing since it seems like the issue is fixed for now.
Resoldering all of the GPS connections seems to have solved it.

closed 2019-11-19 by IVData