iNavFlight/inav #5194: SpeedyBee F4 AIO V2 ESC Protocol Issue

seriousrob, 2019-11-08:
## Current Behavior
Speedybee F4 AIO V2 won't stick to ESC protocol set to DSHOT150 or higher.

## Steps to Reproduce
1. Flash INAV 2.2.1 or 2.3.0
2. Configure the quad with the multirotor guide
3. Enable DSHOT150 or higher
4. Check if ESC protocol reverted back to STANDARD after reboot of FC

## Expected behavior
It should keep the protocol setting, like Betaflight 4.1.0, instead of reverting back.

## Additional context
https://pastebin.com/dtHzn7Ki

---

- FC Board name and vendor: SpeedyBee F4
- INAV version string: 2.2.1

stronnag, 2019-11-08:
Alternatively, one  could check the documentation for boards on which DSHOT is supported, and set the ESC protocol accordingly.

seriousrob, 2019-11-09:
I must have overlooked that point when i checked it earlier. So this bug is not a bug, it should be a feature request.

closed 2019-11-09 by seriousrob