iNavFlight/inav #5198: "Airplane" preset dedicates two outputs to motors, like "Airplane with Differential Thrust"

doughadfield, 2019-11-11:
## Current Behavior
Presets "Airplane" and "Airplane with differential Thrust" seem identical, and both map S1 and S2 outputs to motors. This is an issue for me and, I suspect, everyone who uses flight controllers with limited servo outputs, as I need to reclaim S2 for use with a servo. Most planes have one motor (some of mine are twin, but I often use the same servo channel for both ESCs, if I'm not using differential thrust). I see that in the mixer presets drop-down in Configurator, there are separate presets for "Airplane" and "Airplane with differential thrust" - currently these seem identical! Surely the sensible configuration would be to have a single motor (and servo mappings starting at S2) for the "Airplane" setting, with "Differential Thrust" setting dedicating S1 and S2 to motors, as at present. Would it be possible to add this fix to the development schedule?

## Steps to Reproduce
Mixer section of Configuratior forces S1 and S2 outputs to motors, even when only one motor is configured

## Expected behavior
"Airplane" preset should dedicate only one output (S1) to motor

## Suggested solution(s)
"Airplane" preset should dedicate only one output (S1) to motor, with S2 onwards mapped to servos

## Additional context
Many low-cost flight controllers (the most popular ones folks are using nowadays eg Omnibus F4 Pro etc) have only 4 servo outputs, or in the case of the omnibus, only 4 with all pins available for direct plug-in. Surely the most common fixed-wing configuration with FC would be single motor, Rudder, Aileron, Elevator (not necessarily in that order :-)) so only 4 servo outputs should be required. However current iNAV behaviour un-necessarily forces the need for a fifth servo output.

---

All current and RC versions of iNAV that I've tried (2.0, 2.1, 2.2)

- FC Board name and vendor:
I'm using Omnibus F4 Pro V2 and V3 boards
- INAV version string:
All current and RC versions of iNAV that I've tried (2.0, 2.1, 2.2)

DzikuVx, 2019-11-11:
This topic pop up once a while and no, it's not a bug per se. It is a hardware limitation of majority of flight controllers and STM32 MCUs on them. 
For a longer explanation, watch https://www.youtube.com/watch?v=v4R-pnO4srU

closed 2019-11-11 by DzikuVx

doughadfield, 2019-11-11:
thanks a lot for your very quick reply! - I'll check Pawel's video, but I understand that it's to do with SOC timers. Again, if releasing S2 for servos means locking the motor output to the same timing as the rest of the servos (50Hz frame, etc) then again I think that's a perfectly acceptable behaviour for "Airplane" preset, as almost all airplanes will be using standard ESCs and we don't need the kind of throttle control that Quads require (with all their DSHOT gubbins). So again, if it can be documented that the "Airplane" preset locks all PWM outputs at standard servo frequency, then I'm sure almost ALL fixed-wing pilots will be happy (as after all, that's what all PWM output standard receivers do!).

doughadfield, 2019-11-11:
Sorry, Pawel, I hadn't realised that it was you that replied! :-) - To my question, Is it possible to compile an iNAV executable that is able to switch timer settings according to Configurator preset (rather than have them fixed at compile time)? I know that some folks have compiled custom versions of code to lock the S1 and S2 timer frequencies to 50Hz, so as to use a servo on S2 - is this possible to switch at runtime?

DzikuVx, 2019-11-11:
no it is not possible to switch at runtime. For every board you have to resolve timer issues independently. What Configurator applies is not really relevant. If for example there is only one Motor pin in the target definition, but you configure 7 motors, only one Motor will be working.

doughadfield, 2019-11-11:
I see, thanks for the clarification. So if we wanted single-motor versions of iNAV, they would need to be separate compilations (which I can see would add huge load to the build process for every new version - I was a software developer myself, so I understand the issues :-) - looks like I'll need to do my own compilation, then - and update it with every new release. Never mind, compared with your workload, that's probably quite trivial ;-))
thanks again