iNavFlight/inav #5200: RunCam Device Not Working On Mateksys F405SE (Wing)

CradyLab, 2019-11-11:
## Current Behavior

I have a RunCam Split 2S connected to UART 6 of a Matek F405 Wing and cannot get the FC and camera to communicate.  UART is set to _RunCam Device_ and have tried the associated Modes on several channels without success

Camera communication is set to 'FC communication', not PWM.  Camera TX is connected to FC RX6 pin.  Camera RX is connected to FC TX6 pin 

Tried INAV builds 2.0 through 2.3RC

Split 2S running firmware version 3.2.0

As an aside, I also have an AKK Dominator 2 VTX connected to UART 5 via SmartAudio and that is working as expected (TX stick commands are able to change bands, channels, power etc.)

RX is a FrSky R9 Slim+ in FPORT mode.  Full telemetry and other features/sensors all functioning normally

## Steps to Reproduce

1.  Set UART 6 to RunCam Device
2.  Connect camera TX/RX pins to FC RX/TX pins, respectively
3.  Assign mode CAMERA POWER to RC channel
4.  Toggle CAMERA POWER between 1000us and 2000us

Followed the mfg. instructions [here](https://www.runcam.com/download/split2s/RunCam_Split2s_manual_EN.pdf)

## Expected behavior

Toggling CAMERA POWER between 1000us and 2000us should cause the recording to start/stop

Camera never responds

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:   Matek Systems F405 Wing  +  RunCam Split 2S

- INAV version string:  2.3.0

CradyLab, 2019-11-14:
Seems RunCam has confirmed this is an issue.  I just received the following in an email from RunCam Customer Service: "_From our test, iNAV doesn't work with  our camera any more._"

fiam, 2019-11-14:
I tested this back in 2.1 or so and we haven’t touched the RC Split driver since then. Are you running the latest camera firmware?

fiam, 2019-11-14:
@CradyLab see https://github.com/iNavFlight/inav/pull/4650

CradyLab, 2019-11-14:
I am, thanks.  I made it a point to update the firmware after the first failed test.

CradyLab, 2019-11-14:
> @CradyLab see #4650

I did try different BAUD rates without success.  As I noted, the RunCam team also was unable to get the Split 2S to work with the Matek F405 Wing under iNAV 2.2 and later

LMK if it it would help to post a CLI dump. Thanks

Jetrell, 2019-11-15:
The Runcam protocol has been hit and miss on the boards I have tested.
My F7 boards work fine with Split 1 and 2 series(up to 2.2.1). While some of the F4 boards don't work.

With respect to the Caddx Tarsier UART adapter, that uses the RC protocol to start/stop record and change FPV camera settings. It will not work at all using iNav. But it works fine with BetaFlight.

Cooler044, 2020-01-06:
Just configured runcam split 2 on matek wing f405 ot 5th serial on inav 2.3.
Works out of the box with 57600 baud rate, but sometimes doesnt work after start and I have to replug the battery. Also works after a software restart.
As I read through google, this is the known issue, the camera sometimes takes longer to initialize the port and FC cant see it.

Riffer, 2020-01-06:
I must add myself to the queue, using the RC Split V1 and a Matek F722 I have no success at all.
The previously developed workarounds in #4252 with settings rcdevice_init_attempts = 6, rcdevice_init_attempt_interval = 1000, rcdevice_protocol_version = 1, rcdevice_feature = 7 worked.

Cooler044, 2020-01-07:
@Riffer save and restart from any tab of configurator.

Riffer, 2020-01-11:
Did not change the situation. I will try to go back to the formerly version; may I ask, why the parameters for rcdevice_* are not included into the final branches?

Riffer, 2020-01-12:
I have tested all official releases from 2.1 to version 2.3 and none of them is able to provide the feature. The only working version for me at a Matek F722 and RunCam Split Version 2.1 is this one:

# INAV/MATEKF722 2.1.0 Jan 15 2019 / 18:21:17 (9c12c7682)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

It contains the parameters from the original pull request #4252 from @azolyoung (I do not remember where I got the hex file from, but it has to be from there).
 
These settings are right and work any time directly after the bootup:

rcdevice_init_attempts = 6
rcdevice_init_attempt_interval = 1000
rcdevice_protocol_version = 1,
rcdevice_feature = 7 
 
I have had a look into the sources, but my understanding is limited. I can see that the init attempts and interval are the same there, but I do not understand how the other parameters work or are initialized in the source code at all.

CradyLab, 2020-01-12:
Thanks for the effort, Riffer!  Hopefully the dev team will look into this and get it fixed sooner than later

> I have tested all official releases from 2.1 to version 2.3 and none of them is able to provide the feature. The only working version for me at a Matek F722 and RunCam Split Version 2.1 is this one:
> 
> # INAV/MATEKF722 2.1.0 Jan 15 2019 / 18:21:17 ([9c12c76](https://github.com/iNavFlight/inav/commit/9c12c7682cea6c0e67049db42fc3940b76ce8a14))
> # GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
> It contains the parameters from the original pull request #4252 from @azolyoung (I do not remember where I got the hex file from, but it has to be from there).
> 
> These settings are right and work any time directly after the bootup:
> 
> rcdevice_init_attempts = 6
> rcdevice_init_attempt_interval = 1000
> rcdevice_protocol_version = 1,
> rcdevice_feature = 7
> 
> I have had a look into the sources, but my understanding is limited. I can see that the init attempts and interval are the same there, but I do not understand how the other parameters work or are initialized in the source code at all.

Riffer, 2020-01-13:
How does the sudden close of a still open bug help with solving?

CradyLab, 2020-01-13:
Sorry, that was an accident.  My bad.

Riffer, 2020-01-20:
I did some research on my own. Looking at the C code from today remembers me on my own small projects from 30 years ago - and my knowledge of C is 'a little bit rusty', but somehow it works. 

By cherrypicking 9c12c76 into the current development version and testing around with different parameters and more output (is there no debug console available most of the debug_mode settings are not even documented? I had to extend CLI status to get some variable outputs as a workaround) conclude, that the features are not always received from the RunCam device. Sometimes not even the boot screen is shown while starting up the system (by plugging in the battery).

With the setting for "rcdevice_features" from the patch above the missing response from the device is just being ignored and "isFeatureSupported" is taken from the setting instead. And it works! Most times at least.  

To be fair, the problem seems to be a really unstable implementation in 2.1 of the runcam split V1 at my place. But I am still looking into options to make it work, so please do not close this case.
If you want I can compile a version for your F405SE and you can use "rcdevice_features = 7" to test is with yours, @CradyLab?

CradyLab, 2020-01-20:
Cool.  Yeah I'd love to give it a try - thanks for the efforts, Riffer.  (My own C expertise is far too limited to be of any use here)

Riffer, 2020-01-20:
@CradyLab 
please be aware this is untested software and may contain new or other bugs than expected.
use "get rcdevice" to see the different possibilities.
As mentioned before "rcdevice_feature=7" may enable all features (more or less the button simulation).
Using "status" in cli mode you will see some output about the status of the device, please let me know what you see there. 
[inav_2.3.0_MATEKF405SE.zip](https://github.com/iNavFlight/inav/files/4087400/inav_2.3.0_MATEKF405SE.zip)

CradyLab, 2020-01-20:
Got it - thanks.  I'm fighting off the flu at the moment but I should be able to give it a later in the week.  I'll let you know.  Cheers

CradyLab, 2020-01-27:
@Riffer   Okay.  I uploaded your firmware and after a couple of reboots got the status:

status
RunCam Device:
Features: 0 
isReady: false 
protocol: 1 
System Uptime: 29 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 12.14V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=QMC5883, RANGEFINDER=BENEWAKE, PITOT=MS4525
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=OK, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0x41, 30965760kB, 10/2018, v2.0, 'SD16G'
Filesystem: Ready
Stack size: 6144, Stack address: 0x10010000, Heap available: 1508
I2C Errors: 0, config size: 5767, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 4, cycle time: 1008, PID rate: 992, RX rate: 661, System rate: 10
Arming disabled flags: NAV COMPASS CLI SETTINGFAIL
### ERROR: Invalid setting: rcdevice_init_attempts

#

Riffer, 2020-01-29:
@CradyLab ok, I will redo the compile, a little busy, but hopefully this weekend.

CradyLab, 2020-01-29:
@Riffer Cool.  That makes two of us ; )

wx4cb, 2020-04-04:
any update on this ?

ST02CMC, 2020-06-08:
This seems to be an issue in Inav 2.5.0 as well, I had some Firmware from 2.1.0 that had some code changed to make it work, but it does not seem to have made it into any builds after that, can anyone help with a 2.5.0 build that allows the Runcam Split cam control to work please

Gitfennz, 2022-04-16:
I have had the same issue.  I have had success with the stick commands to change the FPV settings, but no luck with the HD control (On/Off, Record/Stop, Switch Mode) using switches on the radio.  Has this issue been resolved or terminated?  Thanks

INAV 4.1
Radio: Taranis X9D+ (OpenTX 2.3.14)
FC: Matek F405-WSE
Camera: RunCam Hybrid 2 (Firmware 1.1.6)

Thanks

closed 2024-02-02 by DzikuVx