iNavFlight/inav #5202: Range finder and optical flow on iNav on Kakute F7

multinnov, 2019-11-12:
Hi,

I woul'd like to know if it is possible to use a range finder and optical flow on Kakute F7 with iNav?

I try with Pix4flow by Holybro (https://shop.holybro.com/px4flow-kit_p1035.html). But can't work ...

I think so the better in to use this device: https://www.youtube.com/watch?v=XUncioDjs1Q
But it is only a prototype...

Do you have any ideas ? I don't know what can I use... On Kakute we have: UART, I2C

Thank you :)

DzikuVx, 2019-11-25:
Pix4flow is not supported by INAV. INAV supports Matek OpFlow board (upcoming) and Cheerson CX-OF

closed 2019-11-25 by DzikuVx

eremteknoloji, 2020-10-23:
> Pix4flow is not supported by INAV. INAV supports Matek OpFlow board (upcoming) and Cheerson CX-OF

https://tr.aliexpress.com/item/33005508643.html

https://www.banggood.com/Holybro-Kakute-F7-AIO-V1_5-STM32F745-Flight-Controller-w-or-OSD-PDB-Current-Sensor-Barometer-for-RC-Drone-p-1317570. html? akmClientCountry = TR & rmmds = search & cur_warehouse = CN

Hello Pawel, I bought the product on the link. Inav 5.2.2 version is installed on kakute f7. Optical flow option is cx-of. serial port 4 optical flow 19200 is selected. But the optical flow sensor indicator is red. still not seeing. what can i do I will be glad if you help me.