iNavFlight/inav #5203: F7 Kakute V1.5 Reboots after flashing

DoneWorld, 2019-11-12:
## Current Behavior

## Steps to Reproduce

1. DFU Mode Kakute F7 V1.5 30.5 x 30.5 board
2. Download F7 Kakute Latest Stable Release directly from INAV
3. Full erase and flash
4. Reboot the controller disconnect power and power up again

## Expected behavior
After Flashing with INAV the controller reboots continuously.

## Suggested solution(s)
 Check V1.5 Board compatibility

## Additional context

---

- FC Board name and vendor:
- Holybro Kakute F7 V1.5 STM32F745 Flight Controller W/ OSD Barometer

DzikuVx, 2019-11-12:
See #5153

closed 2019-11-12 by DzikuVx

MaxLansing, 2019-11-12:
Well, i can confirm the behaviour. 
KakuteF7 2.3 RC1 stuck in boot-loop.
the only working FW is Pawels 2.2.1 Port @ #5153

DzikuVx, 2019-11-12:
OK, my bad, I forgot that a fix did not went into 2.3RC1. 
Here is the hex file for Kakute F7 of INAV 2.3 with fixed Dynamic Filters as well

[inav_2.3.0_KAKUTEF7.hex.zip](https://github.com/iNavFlight/inav/files/3837748/inav_2.3.0_KAKUTEF7.hex.zip)

MaxLansing, 2019-11-12:
Big thanks to Pawel! Thank you! It boots, settings are restored and the quad flies.
Tested DSHOT 300 & 600, so far all good.
One thing, i can't get BLHeli32 to talk to the ESC anymore. When trying to connect it says:
Initialization of FC denied! Check all Motors stopped!
Booth Versions of BLHeli32 32.7.0.4 & 32.7.0.2