iNavFlight/inav #5209: Optional Betaflight rates

ianrmurphy, 2019-11-16:
## Current Behaviour
iNav in Acro (rate) mode has it's own rate calculation which does not match that of Betaflight.

## Desired Behaviour
There should be an option to specify and use Betaflight rates.

## Suggested Solution
Implement a cli parameter (suggest 'rates_type', allowed values INAV, BETAFLIGHT) to choose between iNav and Betaflight rate calculation of '<axis>_rc_rate' and '<axis>_srate'.

## Who does this impact? Who is this for?
With the recent improvements to iNav perfomance on multirotors in Acro (rate) mode there will be users who wish to switch from Betaflight to iNav. These users may have carefully tuned Betaflight rates which they wish to port over to get a consistent flying experience.

## Additional context
Betaflight cli parameter 'rates_type', allowed values: BETAFLIGHT, RACEFLIGHT, KISS.

ianrmurphy, 2019-11-16:
Having made some small contributions to Betaflight I am happy to work on this feature. Would appreciate some guidance on the iNav features roadmap and iNav pull request etiquette.

DzikuVx, 2019-12-24:
I honestly think Betaflight rates are one of the worse rates ever. Being unitless with this mix of rates, rc rates and super rates it's counterintuitive and hard to setup. INAV deliberately moved out of those into degrees per second

closed 2019-12-24 by DzikuVx

chronowarper, 2023-11-11:
Would you consider adding an option to use the rate curve from this PR on arducopter? 

It uses the Betaflight super rates for the expo calculation and uses deg/s as end-point. Leonard has some analysis of the formula as well. I personally have found it difficult to replicate the same curve with inav with just endpoint and one expo curve.

https://github.com/ArduPilot/ardupilot/pull/18667

With this, it's possible for someone flying acro to keep the same muscle memory cross arducopter/betaflight/inav.