iNavFlight/inav #5212: Feature Request: H7 processor support

wingspinner, 2019-11-19:
H7 FC boards are here and very powerful. They come in n For compatible pin outs which enables leveraging existing F7 board designs. SPRacing has two boards shipping. Th benefits are at least twice the processing performance of the F7 and powerful DSP I nstructions  for better imu and filtering as well as optional two processing cores, more memory, so me versions have a graphic processor as well which could enable the holy grail of OSDs - fully graphical/color without a dedicated coated osd chip. Just adding H7 support would be about the same effort as adding F7 support and would enable others to easily begin experimenting taking advantage of H7 new features. Betaflight/cleanflight now support the SPRacing boards.

DzikuVx, 2019-11-19:
H7 support and new SPRF boards support will not happen without a will of @hydra

hydra, 2019-11-19:
I'm happy to assist!  I gave @digitalentity a run down on what's needed for support, I'll recap here:

* makefile updates for CPU support.
* nucleoh743 target (for devs)
* CPU startup code, linker scripts, etc,
* system drivers (DMA, GPIO, UART, USB, etc)

then for H750 support you also need:
* EXST bootloader system support (mostly makefile changes to inject a hash to the .elf/.bin/.hex files)

then for SPRacingH7EXTREME/ZERO/NANO support you need:
* W25N01G flash support via QSPI.
* Load/save config to external flash instead of CPU flash

All the code is there, it just needs someone to port it from BF.

I am more than happy to assist and provide hardware for anyone willing to undertake this.

wingspinner, 2019-12-27:
So, Hydra says he’ll support. What’s the status?

hydra, 2020-01-20:
I don't think anyway has begun work on this and no-one has requested my assistance to help them.

I get a fair few emails asking for support for the H7 boards, @digitalentity do you have any roadmap for H7 support?

digitalentity, 2020-01-20:
@hydra I was under the impression that somebody is already doing support for this. IIRC you gave me the heads up so I won't be surprised when support for H7 pops up.

If nobody is looking into this, I can have a look if I have a few boards to debug on.

KenImhof, 2020-01-20:
This would be awesome to have this supported in iNav.

hydra, 2020-01-20:
@digitalentity I reasonably sure no-one has started the work yet.

Suggest grabbing yourself a NUCLEO H743 board, there are two variants:

NUCLEO-H743ZI and NUCLEO-H743ZI2

https://www.st.com/en/evaluation-tools/nucleo-h743zi.html

the ZI2 has STLINK V3, the ZI has STLINK V2.

If you give me your address I can get an H7NANO, H7ZERO and H7EXTREME shipped to you.  The latter both have OSD and BMP388, the NANO is more bare-bones, but simpler and very expandable.

You can email me your address or send via Telegram.

Ok?

digitalentity, 2020-02-11:
With 2.4 finalized we can start looking into this

hydra, 2020-02-17:
@digitalentity awesome, let me know if you need anything.

FL125, 2020-02-20:
Hi,
Thank you for the great job you perform.
I would be highly interested in having H7 FCs supported in iNav.

hydra, 2020-04-03:
@digitalentity I'm getting weekly requests for iNAV support the SP Racing H7 boards.  I had two in one day today, and likely that for every user that asks many more don't make the request.  The recent up-surge in requests is probably due to the recent price decrease of the boards.

There's also two more incoming FC's using the H7 that I've designed which use the SP Racing Pixel OSD system - the H7CINE and the H7NP (A.k.a. the new 20x20 mount 'Lightning' board for FlightOne). See https://github.com/betaflight/betaflight/pull/9640 for some target/osd details.

Any update on your progress with porting the H7 support from BF?

digitalentity, 2020-04-03:
It's moving along. Slower than expected, but we'll get there.

DuckitsMick, 2020-04-04:
I have 4 Octo X8 drones grounded in the hangar along with 3 pilots, 4 stewardesses, 5 ground mechanics, and two 360 photographers hotel-ed up on reserve at full pay waiting for iNav to implement this long overdue support for SP Racing H7 flight controllers.  Chop chop Dear iNav Pioneers... once the new H7 code is put into effect then it will be all downhill easy-street clear-sailing from here on out.  It will be a decade before anything more challenging/fulfilling/Zuper comes out than the H7.   Duck!  It's Mick.

rgp1942, 2020-05-01:
The SP Racing H7 boards appear to be the only ones supporting the new BMP388 barometer, which is twice as sensitive as the BMP280 was, making for even more accurate altitude hold.

digitalentity, 2020-06-08:
Looking into H7 support in https://github.com/iNavFlight/inav/pull/5827

However, I don't think we're going to implement H750 support. Minimum [flash size requirement](https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md#new-target-additions) is 512K, while H750 has only 128k which brings us back to F1 era in terms of available flash.

hydra, 2020-06-09:
I can add support for the H750 is pretty easily once the main H7 support is done, as discussed before there three main changes which are well isolated from other code.

1) loading and saving of config to external flash
2) block of ram reserved for running code from.  in BF this is 448K which has been plenty.  Bootloader loads 448 firmware image from external flash into RAM, code executes from RAM (faster too!).
3) minor changes to build system to add a hash to the firmware so the bootloader can verify the contents before it executes it.

I am happy to help with H7 testing and H750 support.

digitalentity, 2020-06-09:
Sounds good, I'll finalize the H743 support and let you take it from there @hydra, SGTY?

hydra, 2020-06-09:
I did some work today on the H7 support, working towards the target definition for the SPRACINGH7EXTREME and getting the build system working better.

WIP branch for H7EXTREME here - https://github.com/spracing/inav/tree/spracingh750-initial-support (doesn't compile yet, early days in iNAV)
PR for some stuff I found here - https://github.com/iNavFlight/inav/pull/5834

hydra, 2020-06-09:
> Sounds good, I'll finalize the H743 support and let you take it from there @hydra, SGTY?

Yes, and I suspect we can do lots at the same time if we co-ordinate a little.  If there's anything you'd like to me look at let me know.  As mentioned in telegram getting a very bare-bones target up and running is a good way to go, then add more functionality as we progress.  once a bare-bones target boots and has MSP available (likely via UART first, since USB is more work) we can likely work on porting features without tripping over one-another.  No need for DMA, PWM, DSHOT, gyros, sensors, spi, i2c, adc, etc straight away.

ghulands, 2020-09-07:
@hydra and @digitalentity  any updates on the progress made?

hydra, 2020-09-07:
@ghulands See #1861 

I'm not sure on the status of things right now, but Matek's H743 board is out now too - http://www.mateksys.com/?p=5159#tab-id-11 and it says 'to be supported soon'.

[image]

countxerox, 2020-09-30:
I take it no news means no progress or just no interest

hydra, 2020-09-30:
@countxerox I wouldn't make that assumption!  I'm super interested and so are many of my customers.  I'm not sure if @digitalentity has had much time to work on H7 support recently.

digitalentity, 2020-10-01:
Oh, there is interest in this. Unfortunately there is only so many hours in a day :smile: 

@hydra correct, recently I have quite little time to work on this. In the next 2 weeks I should have considerably more capacity. This is still on track to 2.6, not sure about production-readiness, but I should get it working soon.

hydra, 2020-10-06:
@digitalentity Awesome.  Do let me know if you need me for some testing/bug fixing/investigation.

SilentKsays, 2021-01-02:
I'm interested to know how H7 support for inav is coming along? Any good news?

d-hartmann, 2021-01-11:
Same here, any updates to the H743 support right now?

denisiussion, 2021-03-22:
Any progress?

denisiussion, 2021-04-02:
Hope H743 support will be in the list of features: iNav 3.0

chr0mcom, 2021-05-26:
time estimate for finishing?

closed 2021-08-23 by DzikuVx

hydra, 2021-08-23:
So it's 100% done now?

DzikuVx, 2021-08-23:
CPU works if you are asking. The only thing that is missing is SDIO driver

denisiussion, 2021-08-23:
and CAN? need it for matek m9n 3100 f4...

DzikuVx, 2021-08-23:
CAN is not H7 specific