iNavFlight/inav #5216: Airbot OMNIBUS F4 Nano v7

gstyle97, 2019-11-21:
## Current Behavior
There is no possibility to use this board with iNav, the v6 is it, and this board is the evolution of v6.
https://store.myairbot.com/flight-controller/omnibusf4nanov7.html

https://www.graupner.de/media/pdf/37/e3/bc/48374-NANOV7_80126-OmnibusF4nanoV7.pdf

## Desired Behavior
Need a target hex file for this board
## Suggested Solution

## Who does this impact? Who is this for?
All the owners of this fantastic board : 
Modifications on v7:

Tantalum Capacitors added for noiseless gyro reading
Solder pads added for SmartAudio and Camera Control (next to Video pads)
FPC port for Frsky RX
This board now has 6 UART ports
PWM pads on the corner
5V BEC, 2A output
Features::

STM32 F405 MCU, Runs Betaflight firmware
ICM Gyro Over SPI Bus
20 x 20 mm Mounting Holes
Supports Lipo direct plugin (3-6S)
Supports 5V 1A BEC output (Buck)
STM32 controls OSD chip over SPI in DMA mode (Betaflight OSD)
More UARTs
25x25mm (20x20mm M3 mount holes)

## Additional context

gstyle97, 2019-11-21:
👍

INAV-FWG-Steve, 2019-12-16:
I second this motion. This would be a great INAV board. Please add a Hex file for this board.

Woga65, 2020-01-14:
@gstyle97 @INAV-FWG-Steve 
Look here: https://github.com/Woga65/inav/releases

Note that LED strip is disabled since i do not have addressable LEDs for testing.
Feel free to drop a not if something is not working.

gstyle97, 2020-01-23:
https://github.com/Woga65/inav/releases

this topic have to be merge with that from @Woga65

closed 2020-01-23 by gstyle97

kapel93, 2020-03-24:
I try to use Omninusf4nanov7 hex. file. Motors did not work. it did not recognize ICM20602. gps, baro, and smart audio work very well. gyro and accelerometer turned MPU6500 in auto mode always.
So, hex.file did not work 100%. If you correct hex.file, it will be great. I like Inav.

INAV-FWG-Steve, 2020-03-25:
Did you enable motors?

kapel93, 2020-03-25:
Dear Steve,
motor configuration was dshot 300, receiver was ok. all modes were ok. All
sensors including external mpu 9250 with baro were ok except magnetometer.
All propeller was out and motors did not work. You said " enabel motors" I
did not see enable choise on the program.
Please send a e mail how to enabel motors.
Regards
Kadir

şunu yazdı:

Woga65, 2020-03-25:
Look for „Enable motor and servo output“. Depending on the INAV version you find it either in the Configuration tab or in the Outputs tab.
Also the Omnibus F4 Nano V7 variants come with no magnetometer and baro, you need to attach external devices to the i2c port.

kapel93, 2020-03-25:
Hi Woga65,
Thank you first of all. I used mixer to activate motors and they worked all. I will test my drone at the weekend and send feedback to you,
[image]
[image]
Regards

Note: I used external Barometer BMP280 by I2C. It worked well.

kapel93, 2020-03-25:
[image]

msyafiq227, 2020-04-07:
I'm try to use this board on flying wing. but the motor won't spin. is there any solution?

kapel93, 2020-04-07:
Sure,
go to output , check motors and servo enable . and go to mixer and choose
your plain and you will see under plain " load ...." check it. you can
choose motor number, flaps , etc. and save all things.
After everythings ok, your motor(s) will work. That is it.
take care
Kadir

8 Nis 2020 Çar 01:23 tarihinde msyafiq227 <notifications@github.com> şunu
yazdı:

chrisdevh, 2023-10-08:
Hi, I hope somebody reads this, I know, old discussion. I found this when searching for a target for the exact board. Airbot Ombibus F4 Nano V7. I tried to follow the provided links, but did not found the according hex-file. Could someone help me out? Thank yo in advance.