iNavFlight/inav #5219: Linear climb and dive for waypoint missions

danarrib, 2019-11-22:
## Current Behavior
When running a mission with different altitude waypoints, INAV will try to climb or dive to the next waypoint altitude as soon as it can, constrained by `nav_auto_climb_rate`, `nav_fw_climb_angle` and `nav_fw_dive_angle` variables.

## Desired Behavior
INAV should save energy by climbing or diving linearly, and not spend a lot of energy to reach that altitude quickly just to do the remaining travel on the waypoint altitude.

## Suggested Solution
If next waypoint is too far, there's no need to make a hard climb or dive to reach the desired altitude as fast as possible. INAV can climb or dive in a linear manner so it will reach the waypoint altitude only when it reaches the waypoint itself. Just like the new RTH_LINEAR_DESCENT mode.

## Who does this impact? Who is this for?
Every INAV user, Fixed Wing or Multirotor, who wants to fly missions with more efficiency.

hali9, 2019-11-26:
I think about something like this:
[image]
Before starting a new waypoint, we remember the last achieve point and the current altitude.
We can't use the altitude from the last achieve  waypoint because it may not be reached.
If the active waypoint is first (enable waypoint mode) then the "last achieve waypoint" will remember the position when the waypoint mode was started.

danarrib, 2019-11-26:
I'm not sure if this suggestions is related to the main topic @hali9 .

INAV doesn't consider altitude when marking a waypoint as "reached". Even if aircraft couldn't get to the waypoint altitude, INAV will consider this waypoint as reached and will proceed to the next waypoint (and will try to reach that wp altitude on the way.

hali9, 2019-11-26:
I will agree with You, that's what I mean.
Lets:
The height at which the plane is when you start WP mode = 10m
Distance to WP1 = 50m
WP1 altitude = 40m
Distance from WP1 to WP2 = 100m
WP2 altitude = 65m

The plane reaches WP1 at say 35m, i.e. it climbed 25m (using maximum climb).
WP1 has been reached (but the altitude of WP1 is not), so we must remember these 35m.
Now the plane flies to WP2, has a distance of 100m and must climb 30m (65m-35m) and it succeeds after about 60m distance. And this is not needed because it can climb slower, but how to calculate this.
We need altitude when WP1 was reached (35m) and not WP1 altitude (40m).

And now from a simple calculation at what altitude the plane should be at each place between WP1 and WP2. So after 60m distance it should be: 35m + 60m/100m  * (65m-35m) = 53m.
As You see the WP1 altitude (40m) is not need for this calculation,
but the altitude when WP1 was reached (35m) is necessary.

danarrib, 2019-11-26:
This is not exactly the way navigation works. Navigation is handled by a loop process. For each iteration of this loop, INAV will recalculate everything from scratch.

In your example: Aircraft is just over the WP1, but at 35m altitude instead of 40m as set in WP1.
Now INAV will try to reach WP2. It'll see that NOW it's at 35 meters altitude, so to reach WP2 altitude it needs to climb 30 meters and move 100 meters forward.

To do it in a linear manner, it needs to maintain a 16,7 deg climb, so it'll reach 65m altitude when it flies 100m forward.

But INAV doesn't need to store the old altitude number... Because on the next loop iteration, everything will be calculated again. Let's say now aircraft advanced 5 meters forward and climbed 2 meters... So current altitude is 37 and there's 95 meters forward and 28 meters upward remaining... Now INAV will try to keep 16,4 climbing angle (instead of the old 16,7 value)... And this calculation will be made hundreds of times per second, all the way to the WP2.

hali9, 2019-11-27:
If you could see: https://github.com/iNavFlight/inav/compare/development...hali9:linear-climb
The distance between WP's is only calculated when loading the active waypoint.
The distance to the active WP what's left is calculated anyway, so we can use it.
We remember how altitude the plane was when loading the WP.
From these 3 variables we can calculate at what altitude the plane should be now.
We set this altitude as desired, the rest will be done by the altitude PID controller.

danarrib, 2019-11-27:
Ok, sorry... Shame on me! You probably know INAV better than me.

I'll remove my unnecessary comments to keep this topic in the main theme.

danarrib, 2020-01-13:
Ok, just to give a feedback. I built and tested the #5292 and #5226 last weekend, on both quadcopter and wing. Working just fine. Flying missions is a lot better now. Thanks a lot @hali9 !

DzikuVx, 2020-02-19:
I belive we can close it

closed 2020-02-19 by DzikuVx