iNavFlight/inav #5221: Support Kakute F7 HDV

sebaszz, 2019-11-25:
## Current Behavior
Hi,

Is the FC kakute F7 HDV (DJI version) supported for Inav? 

I tried to get it working with the latest stable and unstable version firmware.
Flashing went ok but after I couldn't connect anymore to the controller. 
Also after rebooting Inav configurator (2.3.2) and FC.

Flashed to BF, no issues.
Tried again to Inav same connection issues.

DzikuVx, 2019-11-25:
Please try `2.3` that is avalable under "Show unstable releases"

sebaszz, 2019-11-25:
tried a couple of times without succes. Are there other options?

[image]

DzikuVx, 2019-11-26:
I will contact Holybro

myhrmans, 2019-11-27:
Any success on this?

DzikuVx, 2019-11-27:
not really, INAV devs have no hardware to test. You might help by contacting Holybro with a request to help as well

sebaszz, 2019-11-27:
I will contact them

maybe a noob question but can the hardware data not retrieved from the betaflight firmware file?

myhrmans, 2019-11-28:
I also contacted them, they have contacted you iNav so hopefully this can be solved.

wahightower, 2019-12-07:
This does not appear to be a Feature Request rather a bug.  I have the same problem.  Cannot connect Kakute F7 to Inav using Inav configurator 10.6.0 and Kakute F7 target 2.3.0.  No problem connecting to Betaflight.

DzikuVx, 2019-12-07:
HDV is currently not supported. It should be supported in a next few weeks.

wahightower, 2019-12-07:
Thanks for the response.  Look forward to a new target.

TiTTaN40, 2019-12-09:
I have the same problem, I'm looking forward to MAJ

DzikuVx, 2019-12-14:
@TiTTaN40 @wahightower @myhrmans @sebaszz 
[inav_2.3.0_KAKUTEF7HDV.zip](https://github.com/iNavFlight/inav/files/3963958/inav_2.3.0_KAKUTEF7HDV.zip)

The hex file for Kakute F7 HDV is in the attachment. Please check if it's working correctly

wahightower, 2019-12-14:
Connects and configures.  Have not flown yet.  Thanks for the work you do.

TiTTaN40, 2019-12-14:
It's awesome,
Bravo to the programmers.
Someone to test the OSD on DJI Glasses

DzikuVx, 2019-12-15:
@TiTTaN40 on the topic of DJI OSD, nothing changed really

sebaszz, 2019-12-15:
Working for me as well.  haven't flown yet. 

Thanks!

DzikuVx, 2019-12-17:
HDV target is now available in configurator for flashing

closed 2019-12-17 by DzikuVx

cristian893, 2020-01-26:
Hi guys, 
I want to build a custom firmware on Kakute F7 HDV, but I don't have the target files for this version. 
Do you know where I can find them?

Thanks!

giacomo892, 2020-01-26:
Is KAKUTEF7 HDV is just a variant

blaireau2fr, 2020-04-08:
Hi all,
I try to use a KAKUTE F7 HDV on an airplane but I can't flash the firmware (MAC).  Mode DFU is ok, I tried to load firmware online and from local but when I click on "flash firmware" nothing appears... (I have no pb on BF so my cable and drivers should be ok).
Do you have any idea about what I can do to solve my problem?

redjacketKR, 2020-04-20:
Is custom OSD work in new 1.0.0.5 firmware?

myhrmans, 2020-04-20:
@redjacketKR not quite, think it still needs some work. Can write a list later tonight what is working and not

redjacketKR, 2020-04-20:
Thank you for replying. I wanted to move from betaflight due to position/altitude hold and RTL.
But I guess I have to wait a little more.

Some-Won-1, 2022-06-13:
And now... Katute H7 Bluetooth same issue and using Configurator 4.1. Version 5.0 is out, but no files to load into this thing.