iNavFlight/inav #5222: Matek F411 wse FC cannot boot up everytime

bckenneth, 2019-11-26:
## Current Behavior

## Steps to Reproduce

1.Usb plug in not able to boot up FC everytime. 
2.Battery plug in not able to boot up FC everytime
3.Soild Light on all 3 Leds
4. Port found but FC freeze cant connect to Inav  Config

## Expected behavior
boot up normally
## Suggested solution(s)
Keep plug and unplug it. finally it will boot up FC
## Additional context

I tried on Version 2.2.1 all works ok FC boot up everytime. 
I m guess is the OSD causing issue, ONCE I disable OSD features it boot up FC boot up everytime power connect. 
Please help fix it Thanks

- FC Matek F411 WSE
- INAV version string: 2.3 and 2.3 RC3

bckenneth, 2019-12-03:
Any update pls

snowblind64, 2019-12-18:
I am having the exact same issue on the same FC. 2.2.1 boots every time. 2.3 can take a couple tries before the FC will initialize correctly.

bckenneth, 2019-12-18:
> I am having the exact same issue on the same FC. 2.2.1 boots every time. 2.3 can take a couple tries before the FC will initialize correctly.

Looks like no fix yet . Flashed back to stock FW  like 2.2.1.  Friend of my also have this FC. same issue too. Is 100% Bug on FW.

rsilk1949, 2020-01-30:
No problems for me on v2.3.3 powering from battery or USB.

rsilk1949, 2020-01-30:
No problems for me on v2.3. Power from battery or USB.

rsilk1949, 2020-01-30:
No problems for me on MATEK411SE v2.3. Power from battery or USB.

DutchTerror, 2020-02-07:
Are either of you using the 3.3V bus to run any accessories, such as an OLED screen, perhaps?  I found Matek had referred to use the OLED screen on 3.3V, but it would crash the FC.

closed 2020-06-11 by DzikuVx