iNavFlight/inav #5223: RGB LED strip support for Naze 32

jorgebdc, 2019-11-26:
Excuse me for opening this thread,
I readed
https://github.com/iNavFlight/inav/issues/1696
and
https://github.com/iNavFlight/inav/issues/1199
about the problems of some users to make RGB LED strip to work in NAZE32 & iNAV, and about some supposed solutions.
I wanted to make a custom firmware to make RGB LED strip to work. So, I followed all possible indications in the above urls, but I couldn't make it to work.
Of course, I set up properly all the building environment, compliler, etc.
So, I have tried to enable that feature (led strip) in iNav 1.6.1 and iNav 1.7.0 by creating a custom firmware simply by uncommenting the following line:
      #define LED_STRIP
in "/src/main/target/NAZE/target.h" file
In order to free memory I also commented this other
      //#define BLACKBOX
in "/src/main/target/common.h" file
I can build the new firmware, and it apparently works. But when activating the "Multi-color RGB LED strip support" in the "Configuration" page of iNAV Chrome configurator, and saving and rebooting, serial communications crashes and it is impossible for me to connect again to my NAZE32. 
 [image]
So I have to I flash the firmware again to recover serial communications.
The astonishing fact is that I also tested the custom firmware that tonyintn shared:
https://github.com/iNavFlight/inav/files/1009022/inav_1.6.1_NAZE.zip
https://github.com/iNavFlight/inav/files/1008570/inav_1.6.1_NAZE.hex.zip
And I checked that that tonyintn's 1.6.1 custom firmware WORKS!
When saving the LED STRIP activation and rebooting, serial communications apparently crash, but after switching off and on and reconnecting the NAZE32, it works.
So, my question is: what should we do to make a RGB LED strip to work in a custom firmware for NAZE32 and INAV 1.6.1 or any other possible release?
tonyintn's custom firmware is nice, but I would need to activate some other features... 
Thank you in advance

digitalentity, 2019-11-30:
NAZE32 board is no longer supported

closed 2019-11-30 by digitalentity

jorgebdc, 2019-11-30:
Yes, I know. But it is supported until iNav 1.7.3 release.
So I would like to use one of these previous releases..
Thanks,
Jorge

digitalentity, 2019-11-30:
Not supported means no bugfixes or new features. Unfortunately we can't maintain legacy code.
I strongly advise you to get an F4 or F722 board.

jorgebdc, 2019-11-30:
Yes, I know! Thanks, Konstatin,
But I can build old firmware bymyself.
The only issue I would like to know is why once enabled RGB LED Strip in
Chrome configurator my own custom firmware makes a conflict with serial
communications and tonyintn's one works. If I could contact tonyintn it
would very helpful.
Thank you,
Jorge

jorgebdc, 2019-12-01:
So, don't know if @tonyintn is anywhere.
@tonyintn, could you be so kind to give a response?
Thank you in advance,