iNavFlight/inav #5229: F.Port (FPORT) no direct RSSI

kniuk, 2019-11-29:
Mateksys F765-WING (iNAV 2.3.0 final) with FrSky R9MM with 20190221 FLEX FPORT firmware.

PROBLEM: No native RSSI (should be read from FPORT data feed)

Channel 16 is sending RSSI info (which is OK according to frsky  R9MM manual)
Same behaviour with LBT firmware 20190221.

settings used:
rssi_adc_channel = 0
rssi_source = AUTO
rssi_channel = 0
rssi_min = 0
rssi_max = 100

Arvycka, 2019-12-02:
set rssi_source = protocol - that did if for me. I think auto doesn't work for FPort in 2.3.0

kniuk, 2019-12-03:
> set rssi_source = protocol - that did if for me. I think auto doesn't work for FPort in 2.3.0

It works ! However needed to change rssi_max to 50 from 100, because RSSI was shown as divided by 2 (known bug as I was able to read elsewhere)

kniuk, 2019-12-03:
I would suggest two things:
1. try to get RSSI from "protocol" by default if no other stuff selected.
2. find a bug with RSSI divided by 2 and fix it.

Cooler044, 2019-12-05:
Same situation.
R9 slim+, latest flex Fport FW.
Matek F405 wing latest inav
# version
# INAV/MATEKF405SE 2.3.0 Nov 24 2019 / 11:01:22 (d0bdd38c0)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

FPort works great, the only thing is RSSI didnt work with rssi_source set to AUTO.
Had to manually set it to PROTOCOL and now i have RSSI :)

simtcrom, 2019-12-14:
Issue - R-XSR with F.Port not getting RSSI on FC.

FC - Airbot Omnibus F4 Nano V6
INAV Version - 2.3.0
Rx - R-XSR flashed with v191128 (F.Port). Rx wired from inverted hack pad to UART1.

Already tried set rssi_source = PROTOCOL and set rssi_max = 100

dump attached
[dump_inav.txt](https://github.com/iNavFlight/inav/files/3963547/dump_inav.txt)

CrazyFlutter, 2019-12-22:
**Matek F722-STD** (iNAV 2.3.0 final) with FrSky R9MM with FLEX F-PORT firmware.
set rssi_source = PROTOCOL and  set rssi_max = 50 made it works perfectly well :)
With rssi_source on Auto I had no RSSI in my OSD

DzikuVx, 2019-12-24:
I belive it's fixed by #5285 so closing. Thanks

closed 2019-12-24 by DzikuVx