iNavFlight/inav #5232: GPS No data

DJP0P, 2019-11-29:
Hello I have a problem with gps BN-220 it is wired right it lights up it show up blue in INAV but in GPS tab is doesnt show location data cant lock and shows 0 sats in osd. I tryed UBLOX, NMEA, UBLOX7 it is blue on UBLOX7 and UBLOX then I tryed all the baud rates nothing helped. Swapping uarts didnt helped. I have Matek F411-Wing FC
[image]
[image]
[image]

Jetrell, 2019-11-29:
@DJP0P It has plenty of `Timeouts`.  Do you have another GPS unit to test it with?

DJP0P, 2019-11-29:
Defective GPS unit

closed 2019-11-29 by DJP0P