iNavFlight/inav #5233: omnibus f4 ledstripm5 leds not working

JurikRistjouw, 2019-11-30:
## Current Behavior
leds don't work

## Steps to Reproduce
flash 2.3.0 

## Expected behavior
leds that work 

## Additional context
# diff

# version
# INAV/OMNIBUSF4PRO_LEDSTRIPM5 2.3.0 Nov  6 2019 / 19:49:08 (567787ef7)
# GCC-8.3.1 20190703 (release) [gcc-8-branch revision 273027]

# start the command batch
batch start

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix 0 0 11 100 0 -1

# servo

# logic

# gf

# feature
feature DYNAMIC_FILTERS
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature RSSI_ADC
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 0 64 115200 115200 0 115200
serial 5 2 115200 115200 0 115200

# led
led 0 11,10:S:FW:0
led 1 12,10:S:F:0
led 2 13,10:S:FN:0
led 3 6,10:S:FW:0
led 4 5,10:S:F:0
led 5 4,10:S:FN:0

# color

# mode_color

# aux
aux 0 0 1 1800 2100
aux 1 1 0 1125 1275
aux 2 2 0 1275 1500
aux 3 4 2 900 2100
aux 4 3 2 1400 2100
aux 5 9 2 1400 1650
aux 6 8 4 1925 2100
aux 7 19 3 1925 2100
aux 8 21 4 900 1150
aux 9 17 1 1800 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 0 0 V
osd_layout 0 1 0 12 H
osd_layout 0 3 8 6 H
osd_layout 0 7 13 13 H
osd_layout 0 9 11 14 H
osd_layout 0 11 0 14 V
osd_layout 0 12 5 14 V
osd_layout 0 13 22 7 V
osd_layout 0 14 3 0 V
osd_layout 0 15 3 7 V
osd_layout 0 22 23 0 V
osd_layout 0 23 24 0 V
osd_layout 0 26 26 1 H
osd_layout 0 28 23 14 V
osd_layout 0 30 2 13 V
osd_layout 0 32 0 14 H
osd_layout 0 34 10 0 V
osd_layout 0 40 19 0 H
osd_layout 0 54 0 13 V
osd_layout 0 96 19 1 H
osd_layout 0 97 10 14 V

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 100
set gyro_lpf_type = PT1
set gyro_stage2_lowpass_hz = 200
set acc_hardware = MPU6000
set acczero_x = 54
set acczero_y = -4
set acczero_z = 157
set accgain_x = 4099
set accgain_y = 4058
set accgain_z = 4003
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set mag_declination = 150
set magzero_x = 122
set magzero_y = -101
set magzero_z = -131
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = FPORT
set motor_pwm_rate = 4000
set motor_pwm_protocol = MULTISHOT
set failsafe_delay = 9
set failsafe_procedure = RTH
set bat_voltage_src = SAG_COMP
set yaw_jump_prevention_limit = 200
set model_preview_type = 3
set applied_defaults = 2
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_min_sats = 8
set deadband = 1
set yaw_deadband = 4
set mc_airmode_type = THROTTLE_THRESHOLD
set nav_auto_climb_rate = 300
set nav_manual_speed = 1000
set nav_manual_climb_rate = 1000
set nav_landing_speed = 300
set nav_land_slowdown_minalt = 300
set nav_land_slowdown_maxalt = 500
set nav_rth_climb_first = OFF
set nav_rth_alt_mode = CURRENT
set osd_video_system = PAL
set osd_left_sidebar_scroll = ALTITUDE
set osd_right_sidebar_scroll = GROUND_SPEED
set osd_sidebar_scroll_arrows = ON
set name = DiatoneGPS

# profile
profile 1

set mc_p_pitch = 44
set mc_i_pitch = 60
set mc_d_pitch = 25
set mc_i_roll = 50
set mc_d_roll = 25
set mc_p_yaw = 45
set mc_i_yaw = 70
set dterm_lpf_hz = 90
set use_dterm_fir_filter = OFF
set yaw_p_limit = 300
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

set battery_capacity = 1500
set battery_capacity_warning = 225
set battery_capacity_critical = 150

- FC Board name and vendor:# omnibus f4pro
- INAV version string: INAV/OMNIBUSF4PRO_LEDSTRIPM5 2.3.0 Nov  6 2019 / 19:49:08 (567787ef7)
# GCC-8.3.1 20190703 (release) [gcc-8-branch revision 273027]

Thanx in advance!

Jurik

JurikRistjouw, 2019-11-30:
2.2.1 fine, 2.3.0rc1 fine (but baro not working) from thereon rc2 and 2.3.0 leds fail.

Andrey-Prikupets, 2019-12-01:
Same with me. 2.2.1 = LEDs on M5 work fine; 2.3.0 = LEDs on M5 do not work;

tiriad, 2019-12-01:
I've been looking at the code and in the last changes has been ignored a line that influences the behavior of the led, in this compilation should be corrected, please try the rest of features if they could be affected, if it works correctly for the next version the change will be included 

[inav_2.3.0_OMNIBUSF4PRO_LEDSTRIPM5.zip](https://github.com/iNavFlight/inav/files/3908313/inav_2.3.0_OMNIBUSF4PRO_LEDSTRIPM5.zip)

JurikRistjouw, 2019-12-01:
@tiriad you sir, are a miracle worker! Thank you. Leds are working again!

LawrenceLDJr, 2020-01-14:
Omnibus F4 V3 has this problem too.

PR #5238 (2020-01-22): Led fix in omnibusf4pro_ledstripm5 and simplified the code
Led fix #5233 in omnibusf4pro_ledstripm5 and simplified the code for the other targets based on this target

I have not taken into account the first line in past revisions where omnibusf4pro_ledstripm5 was defined as omnibusf4pro, seeing that this form is simpler than others also I have applied it to the variants of omnibusf4v3.

>
closed 2021-07-28 by DzikuVx