iNavFlight/inav #5237: Save to remap UART to SoftSerial?

UnchartedBull, 2019-12-01:
Hi,

I haven't found an issue covering this ... I have a Corner F4 Nano (basically the 20mm version of the F4 Corner, and probably a clone). I want to hook up a BN-180 GPS (connected to UART6), SmartAudio (connected to UART1 TX) as well as a Telemetry FrSky Receiver. SBus is connected to UART3 RX and S.Port to UART3 TX.

I can't use F.Port because it is an F4 chip with only One Way Inversion.

I had a look into the target file for OMNIBUSF4 and was wondering whether I could manually remap UART3 TX to SoftSerial TX in order to get SoftSerial working on UART3 TX. Here are my changes: https://github.com/UnchartedBull/inav/commit/1d1d6c5ca2c27cef5bc65ad860033900875a7db3. I understand, that this is very specific for my Use Case and disables the normal UART3 TX, but I don't need it anyways.

I just heard that there could be some issues with PWM timing, due to servo outputs. But I wasn't sure, whether those apply to remapping of UARTs as well? So my question basically is, whether it is safe to manually build and flash a Firmware with a UART port remapped to SoftSerial.

Thanks!

tiriad, 2019-12-01:
i think that those changes are ok, the timing theme of the servos is that it is very difficult to implement a remap in the cli

UnchartedBull, 2019-12-01:
Ok great, thanks! I'll give it a try and update this issue with my findings. But closing it now, since it is answered :)

closed 2019-12-01 by UnchartedBull

tiriad, 2019-12-01:
I forgot in target.c you must edit this line, the timer:
DEF_TIM(TIM2,  CH3, PB10,  TIM_USE_ANY,                 0, 0), // S6_IN // pad labelled CH6 on OMNIBUSF4PRO

UnchartedBull, 2019-12-01:
Thanks! I‘ll make sure to try this later. If this works I‘m finally switching over to iNav for my GPS Quads.