iNavFlight/inav #5239: GPS working in BetaFlight, not in iNav

UnchartedBull, 2019-12-01:
## Current Behavior
I use an F4 Corner Nano, which is flashed with the iNav OMNIBUSF4PRO Target (OMNIBUSF4SD for BetaFlight). I connected my Beintan BN-180 GPS to UART6 of that Flight Controller. If I enable the GPS feature and assign it to UART6 I get no messages in the GPS Tab (I tried UBLOX, UBLOX7 and NMEA Protocols). If I flash Betaflight and enable GPS on UART6 everything works just as expected.

I also observed that the LED is blinking fast in BF (the supposed 5Hz for the TX LED), while it is blinking slow in iNav (around 1Hz I think). So it probably does not get configured correctly.

## Steps to Reproduce
1. Connect BN-180 to UART6 on OMNIBU F4 Corner Nano 
2. Enable GPS

## Expected behavior
GPS should work and send messages.

## Suggested solution(s)
No idea, if you need any more info let me know.

## Additional context
I also tried this with 1.9 with the exact same behaviour.

https://pastebin.com/Lup4EMKk

- FC Board name and vendor: Corner F4 Nano
- INAV version string: 2.3.0

tiriad, 2019-12-01:
Try setting gps baudrate higher to 115200, gps ground assistance to disabled, and remember that inav without compass gps navegation doesn't work

UnchartedBull, 2019-12-01:
Just tried that and nothing has changed, still a slow blinking TX LED and no receiving messages ...

Navigation just means Waypoints right? RTH and Position Hold should work, don't they?

stronnag, 2019-12-01:
Navigation means navigation, WP, RTH, PH, Cruise at least. All require GPS.

iNav requires bidirectional communications with the GPS (it checks responses to configuration commands), BF may not. Ensure you have RX and TX connected.

UnchartedBull, 2019-12-01:
Yeah GPS makes sense. The Compass thing through me off, as I though iNav can calculate that based on GPS History, just not as accurate.

I double checked that. And BetaFlight receives information about satellites, so this indicates that the RX Line is working. And because BetaFlight can send something to the GPS unit to make it blink faster the TX line works as well. This is at least my understanding. Is there somekind of Debugging so I can view the messages sent and received in the CLI by any chance? The FC doesn't have a BlackBox

DennyRowl, 2019-12-04:
I think this bug is fixed now but Ublox protocol was pointing to NMEA

UnchartedBull, 2019-12-04:
Ok, is the fix already merged into master?  I may have some time on friday to build the latest version from master and try it on the Quad.

closed 2019-12-24 by DzikuVx

Ozyris, 2020-04-13:
https://github.com/betaflight/betaflight/wiki/Board---OMNIBUS-F4-Pro-Corners

This board have bidirectional inverters, betaflight deal with it, Inav not.

UnchartedBull, 2020-04-13:
Well that explains that ... Thanks for the info! I've given up iNav for Quads though, too much of a hassle for having waypoint navigation, which I might use once or twice. I'm still using it for the Wings, but quads are so much easier to setup with BF.

Ozyris, 2020-04-14:
Proper target for this board will be OMNIBUSF4V3. This FW uses PC8 and PC9 as inverter pins.