iNavFlight/inav #5247: DJI Custom OSD Integration in iNav

MadRC, 2019-12-03:
## Current Behavior
DJI have updated the Digital FPV system to support custom OSD over MSP with Betaflight, this allows users to select unto 24 OSD items and place them on screen via the Betaflight configurator.  This does not work in iNav. 

## Desired Behavior
Support for the new Custom OSD in iNav so that users can have same or similar functional in iNAV as Betaflight, allow custom OSD to show and allow configuration in iNav Configurator. 

Support for PID Tuning via DJI FPV system as well.

## Suggested Solution
iNav to add support, currently its not widely known the reason doe lack of compatibly, I totally understand this takes dev time so perhaps users to help fund iNav developers for hardware for adding DJI Custom OSD support.  I my self are willing to test software with my system to help make this happen. 

## Who does this impact? Who is this for?
This will affect fixed wing, quad and other users as Digital is becoming more wide spread daily and the drawbacks are becoming less and less.  Fatshark also have Byte Frost in Beta and this will be using a similar system too for custom OSD interface over MSP. 

## Additional context
None

DzikuVx, 2019-12-03:
I'm affraid this is not working like that. INAV can do nothing to support DJI OSD. DJI OSD has to support INAV, not the other way around. INAV already provides API to work with settings and OSD, just like Betaflight does.

To make this work, DJI has to know there is a need and users want to have INAV support in DJI OSD.  

This is why, my suggestion is to contact DJI support and request INAV support over there.
The link to DJI Support is here: https://www.dji.com/support/product/fpv

Schlumie, 2019-12-03:
Is DJI not just using MSP_OSD_CHAR_READ or similar? This should work with INAV? BattVoltage works with INav so maybe just a MSP cmd is missing/different? Someone has to sniff the used MSP cmds?

DzikuVx, 2019-12-03:
Not all MSP frames are the same between Beta and INAV and we have different OSD elements.
DJI does firmware version family and version check. Yesterday there was an experiment what woujld happen if INAV introduced itself as BTFL to DJI FPV. And it worked.
Still, DJI has to allow to use INAV with this function

Schlumie, 2019-12-03:
What worked? OSD too?

digitalentity, 2019-12-03:
There's another bug waiting until somebody from DJI can chime in #5106

DzikuVx, 2019-12-03:
@dji-sdk ?

SystemsGuy, 2019-12-04:
What prevents us for advertising a supported family and version until such time as DJI responds?

Schlumie, 2019-12-06:
Is there a patch available to test what happens when inav advertise as BT ?

OlivierC-FR, 2019-12-07:
> 
> 
> Is there a patch available to test what happens when inav advertise as BT ?

We made a quick test few days ago, I made an INAV build that presents itself as Host=BTFL  and version =4.1.0 over MSP, didn't seem to work, a user on the INAV telegram chat tested on his goggles (I don't have the DJI goggle myself, does not help with investigations). More tests would be needed, I had a look in the MSP differences between 1.43 (BF) and 2 (INAV) and didn't notice anything too different for the basic OSD items and their positionning, but... short of knowing what the DJI code expect, it's not that simple to find.

Schlumie, 2019-12-07:
I could sniff a Airunit/BF communication to debug the needed MSP telegrams...

idavidstory, 2019-12-08:
I have made many changes trying to get this to work. (Quack like BetaFlight)
MSP_API_VERSION - 0, 1, 42
MSP_FC_VARIANT - BTFL
MSP_FC_VERSION - 4, 1, 1
MSP_OSD_CONFIG - 168 bytes of binary data (compatible with BTFL)
... it is still not working.

Noticed the MSP_STATUS and MSP_ANALOG are different too.

malarcky, 2019-12-10:
I have the system and I can make test if necessary, but dont know how to code.

uotw, 2019-12-11:
I’m getting ready to purchase this DJI system and throw it in a fixed wing. I'm a coder and will help any way I can. If I get a Matek board with iNav, would this be a good test platform?  Also, if I were to send my DJI hardware to one of the iNav developers, how much would this process be accelerated?

idavidstory, 2019-12-23:
Look at the wiki page, associated with this PR
https://github.com/idavidstory/inav/wiki/Hack-to-support-the-OSD-on-DJI's-HD-digital-FPV-system.

digitalentity, 2019-12-24:
Hardware on the way. I'm taking ownership of this feature request. Code for full support for OSD and PID configuration is already in the works.

digitalentity, 2020-01-22:
Closing. Tuning support will come in scope of #5341

closed 2020-01-22 by digitalentity

Thunfisch2, 2020-04-14:
DJI added some new OSD elements in their new firmware which has been released yesterday:
https://forum.dji.com/forum.php?mod=viewthread&tid=212637&extra=page%3D1%26filter%3Dtypeid%26typeid%3D708%26typeid%3D708
"Added new custom on-screen display information: home point direction, home point distance, altitude, and GPS speed"
Today I configured iNav 2.4 with my new DJI air unit and was able to see the home point direction and distance on the left side of the screen, but this is not configured in the OSD menu in iNav. I'm also not able to move the information to a new location. Altitude is available and can be moved. GPS speed is currently not displayed.

myhrmans, 2020-04-14:
Experiencing the same thing as @Thunfisch2 
Maybe reopen this issue?

digitalentity, 2020-04-15:
@Thunfisch2 please open a new issue so we can track this properly towards 2.5

JagerBombastic, 2020-04-21:
I have just tested the new OSD elements (home direction, home distance, GPS speed and altitude) and the seem to function as intended . I can see them and they move them around just fine though I am yet to fly so i don't know if they actually display correct values.

One thing to note was that the before the GPS was configured I noticed that both the home direction and distance were present on the osd and i couldn't turn off or move them as at that point there were no GPS OSD options.

Tested with a Matek F722-SE & Caddx Vista on the 2.4 release if that has any bearing.

digitalentity, 2020-04-21:
@JagerBombastic aha, so there is a bug with displaying OSD items. Makes sense, built-in OSD checks for sensor presence and hides the items even if they are set to "visible" in configuration. DJI has no  idea about available sensors. Thanks for checking, I'll fix this.

FPVZaphod, 2020-10-24:
I realized, in the Configurator the position of the items are very much different from the positions shown in the DJI OSD. Is it possible to fix this, or at least make it a bit better? To make the position shown in the configurator similar to what you get in the DJI OSD?
I am currently playing with a lightweight analog SD recorder recording the analog OSD (with no Cam installed!) from my Furious F35 Flight controller, plus recording the HD Video in the Air unit. After the Flight I add the OSD clip from the SD-Recorder over the clip of the air unit.
It's quite difficult to find a OSD-Setting, which looks nice in the analog OSD, while at least remaining barely usable in the DJI-OSD. Perfect would be to have WYSIWYG in both OSDs, analog and DJI.