iNavFlight/inav #5250: Single sensor pitot tube

FelixvdDonk, 2019-12-03:
## Current Situation

- If a user would like to have airspeed measurement the user needs to buy one of the commonly available airspeed sensors. These consist out of a pitot tube with a differential pressure sensor, attached via silicone tubing usually.

[image]

These types of airspeed sensors can be found in both the digital and analog variants. The all however have a couple of things in common. 

- They are expensive IMO
- They are heavy, especially for smaller airplanes

## Desired Situation

- Airspeed measurement is very useful, even for smaller airplanes. I'd like to see a situation where adding an airspeed sensor is considerably more appealing from both a cost and effort perspective.

## Suggested Solution

According to the formula for airspeed:
[image]

We need ambient air pressure and the stagnation pressure. Given we have a barometer built in most of our flight controllers we know Ps (if the airplane has big air intakes and little exhaust in the fuselage this will not reflect Ps). 

Now then the only thing we need to measure is Pt. For this we could use a basic MEMS barometer chip like thee BMP280 which can be had for roughly 4$ on a breakout board. In a construction as shown below this can measure the stagnation pressure when mounted on the nose of the airplane: 
[image]

Now i hear you think, i2c address conflict! Yes, but luckily we can change the address of the BMP280 to 0x67 or 0x66 by pulling on of the pins to ground. This is shown as jumper you can solder in most of the breakout boards.

**Now for the actual request**

We need the firmware to solve the equation above to get the airspeed. So INAV should give the option to create a airspeed sensor out of an additional I2c pressure sensor. 

This setup also needs to be callibrated with gps speed on a calm day, as the stagnation pressure measured is dependent on the diameter and length of the pitot tube. A callibration mode would need to be added for this I assume.

## Who does this impact? Who is this for?

Fixed wing userbase 

## Additional context

- The max pressure for a BMP280 is listed as 1100HPA in the datasheet, I believe this is enough for it to not saturate during flight but this will have to be tested.

flounderscore, 2019-12-04:
If you plug in the numbers, you'll see that a typical air speed of 30 mph gives you only about 1 hPa dynamic pressure. The data sheet cites 2 Pa resolution, but is the the static pressure stable to that precision? Curious to hear how it works out in practice.

DzikuVx, 2019-12-04:
@giacomo892 IIRC you were experimenting with it some time ago. How it ended?

giacomo892, 2019-12-04:
@DzikuVx I did some experiments.
I had a little rubber tube sealed around a BMP280, after the initial calibration, if you started blowing constantly inside the tube indeed it was showing a pretty constant "air speed".
The issues I had where

- Never found a proper way to seal the sensor and the tube so some air was leaking near the sensor, but indeed the 3d printed solution is nice!
- The BMP280 was stop responding if I blow too hard into it

So I paused the project.

But now that we have way more accurate barometers like the the SPL06-001 it might worth giving another try to it!

FelixvdDonk, 2019-12-04:
@giacomo892  My plan was to do a test with a Lora equipped arduino and two BMP280's first to see what filters need to be enabled/disabled on the pressure sensor. Do you already have some recommendations for whether or not to enable the Infinite response filter? And oversampling? Also if you have some custom firmware already for testing let me know. I'll gladly test it. 

@flounderscore  I believe this dynamic pressure ratio can be made more extreme by making the air intake cone shaped, tapering down to a lower diameter tube running up to the pressure sensor

giacomo892, 2019-12-04:
@FelixvdDonk my firmware was just a little code on arduino to read the pressure and display airspeed using the provided formula, nothing that cannot be done in 10 minutes :) 
Use some filtration of the value, you will notice speed varying a lot otherwise.
You can test it also by attaching it your car with a little oled screen and while driving on resonably flat road (assuming you calibrate your static pressure before moving) you should get airspeed matching the car speed. 
Let me know!

Cris-77, 2020-05-22:
@FelixvdDonk I have read your concept description. You want to measure speed as computation of dynamic speed (from own pitot) and 'static' from the FC board. But FC measure **not static** but **Hydrostatic** pressure. 
Following Bernoulli principles velocity is related to the difference of static(drop) and dynamic(increase) inflow. If it worked, aircrafts with unpressured cabins would use this. Some gliders have static pressure input, not in pitot but on the fuselage in aerodynamically design selected points. 
It will be nice to know your reference to this.

gioclava, 2021-03-12:
I had a similar idea and also read this thread. I have started a project here for a pcb with two baros and an atmega328 to communicate directly the airspeed to inav via msp. Suggestions are welcome.
https://hackaday.io/project/178202-cheap-airspeed-sensor-for-inav

gioclava, 2021-03-31:
I received the PCB and wrote the code for it, first results are promising.

First results:
- The PCB costed me around 5€, components are around 3€ per sensor
- I can detect variations of 1Pa and overtime I get a drift of 10Pa which is not bad as it means that reading above 4m/s 15km/h should be accurate and get more accurate as speed increase

Now I need to translate the readings in MSP to make it compatible with INAV. If someone wants to help: https://github.com/gioclava/barometer/blob/master/src/main.cpp

gioclava, 2021-04-02:
Finally talking to INAV and seeing data on the configurator :) next step 3d printing a case for it.
More info here: https://hackaday.io/project/178202-cheap-airspeed-sensor-for-inav

closed 2022-09-03 by DzikuVx