iNavFlight/inav #5251: Surface, Althold and Poshold in Inav with Matek  Optical Flow & Lidar Sensor 3901-L0X

multinnov, 2019-12-03:
Hi,

I use Kakute F7 with INAV 2.3.0 and Matek device. Surface mode work in flight, but don’t work « flow » operation with flight mode NAV ALTHOLD + SURFACE (I think is normally, because we need to use POSHOLD I think so...)

The sensors are OK, and no problem to set them up.

I added NAV POSHOLD in my settings , and i can’t arm my UAV. I think so POSHOLD is the  reason, because i need to use 2D and no 3D positionning.

How do to fix this problem ? Because in the future, all boards will work with INAV 2.3.0 and + (Because i saw in the past, in Inav 2.0 there went POSHOLD 2D and POSHOLD3D, i 'm not sure...).

Other question : It is not possible to switch in classic ALTHOLD with barometer above 1m50 ? I try with one switch to pass SURFACE + ALTHOLD mode in ALTHOLD mode only, and don’t work… It stay in SURFACE mode…

Thank you

closed 2020-02-15 by stale[bot]

pDR8iLQ4go, 2020-04-10:
I‘m testing this sensor on my hexacopter with the Matek F405 CTR flight controller. SURFACE mode works in combination with POSHOLD. But the copter is flying at a maximal altitude of about 1m. I‘m able to decrease altitude (throttle stick command). But increasing altitude more than 1m is not possible. Is this the expected behavior?

pDR8iLQ4go, 2020-04-10:
Sorry, the answer to my question and the explanation can be found here: https://github.com/iNavFlight/inav/issues/5263