iNavFlight/inav #5252: Can't Calibrate  Accelerometer

uakc, 2019-12-04:
Can't Calibrate  Accelerometer after 3rd step setup keep on reverting back to 1st step.
my board Matek F405-STD
https://youtu.be/AOEnNNUL19A

whoim2, 2019-12-04:
Try reflash with full erase chip

uakc, 2019-12-04:
tried that a few times.. no luck.. maybe F405-STD not supported..

whoim2, 2019-12-04:
> tried that a few times.. no luck.. maybe F405-STD not supported..

Supported 100%, my friend use 2.2.1 inav on this FC and have gps/compass, worked perfectly.
Try calibrate from sticks.

uakc, 2019-12-04:
what do you mean "Try calibrate from sticks. "  how ?

whoim2, 2019-12-04:
> what do you mean "Try calibrate from sticks. " how ?

[image]

whoim2, 2019-12-04:
Also, you may trying calibrate from speedybee software on android

closed 2020-02-27 by DzikuVx