iNavFlight/inav #5259: RECOMMENDED PIDS FOR LARGE HEX?

CradyLab, 2019-12-08:
I replaced the aging Pixhawk on my 790mm hexacopter with a Matek F765-WING and was wondering if anyone had some tips for the PID settings in iNAV (the largest prop size mentioned in the configurator 12 inches and I'm running 15-inch props)

Thanks!

Vissie26, 2019-12-09:
Dear Cradylab,

I am flying a 690 size hexa and my quite good flying pids are:

Roll :  P 155 I 35 D 45
Pitch: P 165 I 35 D 65
Yaw:  P 145 I 55 D 0

I am currently on version 2.2.1

This should fly reasonable for a first try.

I hope you find this info useful.
[image]

CradyLab, 2019-12-09:
Sweet!  Thanks Vissie!  Much appreciated

pDR8iLQ4go, 2020-01-19:
@Vissie26: May you please post your complete dump? I have a 650 size hexacopter and would like to compare the parameterization. Which kind of props and motors do you use?

Vissie26, 2020-01-19:
here you go. 

[hexa dump 2.3.0.txt](https://github.com/iNavFlight/inav/files/4083095/hexa.dump.2.3.0.txt)

Props: https://hobbyking.com/en_us/multirotor-carbon-fiber-t-style-propeller-12x5-5-black-cw-ccw-2pcs.html

Motors: https://hobbyking.com/en_us/turnigy-multistar-4225-610kv-16pole-multi-rotor-outrunner.html

Let me know if you need more info.

pDR8iLQ4go, 2020-01-19:
I'm using a Matek F405 CTR and cheaper stuff from AliExpress:

Gartt ML 3508 700KV Motors: https://de.aliexpress.com/item/32751865371.html?spm=a2g0o.productlist.0.0.ebc25ddaBaUuAo&algo_pvid=e105fdd5-14fa-4e8c-9c03-5c23b005f712&algo_expid=e105fdd5-14fa-4e8c-9c03-5c23b005f712-0&btsid=42319929-d8ec-4b3b-a563-b61a6b45168f&ws_ab_test=searchweb0_0,searchweb201602_9,searchweb201603_55

12x55 Carbon Props: https://de.aliexpress.com/item/32917571592.html?spm=a2g0s.8937460.0.0.3bc22e0ehCftce

Vissie26, 2020-01-19:
That combination should work in my opinion. What kind of battery are you using? I'm flying with a 4 cell 10.000 mAh.

pDR8iLQ4go, 2020-01-19:
Yes, if works fine. But I am using an older Version of INAV, here is the dump: <how can I upload a text file>

I have a 4S 5500mAh, flight time is about 15min.

Vissie26, 2020-01-19:
if you drag the file onto the screen where you can reply it adds it to your message.

Vissie26, 2020-01-19:
your D values for pitch and roll seem a bit high. I had mine up that high and it gave worse results the higher i set them. My current values ar around 60 and a P gain of 140-160 but my frame is 690mm.
You could try (P 120 I 35 D 55) for roll and (P 130 I 35 D 60) and see how it flies.

pDR8iLQ4go, 2020-01-19:
I think, that I‘m going to update INAV first. Getting a sufficient filter setup was a nightmare. I hope that the new dynamic filters do a good job. On my smaller Q380 size frame, if seems to work.

Vissie26, 2020-01-19:
If you activate dynamic filters the video from Paweł Spychalski might be useful if you haven't seen it already.
https://www.youtube.com/watch?v=2YFkDsnPnqE

pDR8iLQ4go, 2020-01-19:
I know this video. Pawel told me, that the old not filter have to be deactivated, in case of using dynamic filters. But in your dump, the notch filters are still parameterized. But you use the dynamic filters as well. Doesn’t that work in combination with the old notch filters?

Vissie26, 2020-01-19:
as far as I know the old values are showimg but not used by the flightcontroller. I have followed pawels video step by step and he doesn't say anything about removing the old values.

pDR8iLQ4go, 2020-01-19:
Yes, nowhere is explained what to do with the old/existing parameters. Maybe they will write it down somewhen.

Vissie26, 2020-01-19:
I'll send pawel a message asking about this. curious about what he has to say about this.

pDR8iLQ4go, 2020-01-19:
Keep me informed. See you...

pDR8iLQ4go, 2020-01-20:
I am using high D values to improve the stability when going down in the copters own turbulence (= propwhash?). But as mentioned, I’ll update INAV and try your D term.

Vissie26, 2020-01-21:
I just got a reply from pawel.

''You might try disabling static gyro notches but one by one and verify
if motors are not overheating''

looks like we have to check blackbox data to verify if the dynamic filters are working or not.

pDR8iLQ4go, 2020-01-24:
Have you planned to disable the „old“ (static) notch filters? If yes, please inform us regarding the results.

Vissie26, 2020-01-24:
I am planning to disable them but the weather forecast isn't too good for the following days. I will report back when the weather gets better (less wind)

Vissie26, 2020-03-08:
Just flown without static notch filters and it seems fine (No hot motors and weird sounds).  

What I did was:

set gyro_notch1_hz = 0
Allowed range: 0 - 500

set gyro_notch1_cutoff = 1
Allowed range: 1 - 500

set gyro_notch2_hz = 0
Allowed range: 0 - 500

set gyro_notch2_cutoff = 1
Allowed range: 1 - 500

set dyn_notch range = LOW
Allowed range: HIGH MEDIUM LOW
 
set dyn_notch_min_hz = 75
Allowed range: 60 - 1000

I hope this is enough info.

pDR8iLQ4go, 2020-03-14:
I've been updating my hexacopter as well. Still flying. Motors don't get warm at all.

@Vissie26: May you please post your dump once again?

Vissie26, 2020-03-15:
I have upgraded to 2.4.0 in the meantime.

This is my current dump file:

[dump 2.4.0.txt](https://github.com/iNavFlight/inav/files/4334288/dump.2.4.0.txt)

pDR8iLQ4go, 2020-03-15:
Most important settings are more or less the same:
[Compare.pdf](https://github.com/iNavFlight/inav/files/4334306/Compare.pdf)

I like "set inav_reset_home = EACH_ARM" to ensure that the home position is where I armed last time. And "set debug_mode = GYRO" to be able to use the Plasmatree's PID-Analyzer.

Vissie26, 2020-03-15:
The reason I set ''set inav_reset_home = FIRST ARM'' is that if i accidentally flip the arm switch in flight, the home position isn't reset. Quite a rare occasion but i'd rather be safe than sorry.

Saw the debug mode parameter also. Was wondering how to use it exactly. Not much info on the wiki.

Would you be so kind to explain a bit about the function if you can? especially VIBE and Gyro?

pDR8iLQ4go, 2020-03-15:
I'm not a real expert in that. But "set debug_mode = GYRO" seems to add some more details to the blackbox file. This setting is necessarry using Plasmatree's PID-Analyzer. See https://www.youtube.com/watch?v=VaqlLSQlJ4M and this hint:
[image]

closed 2020-05-27 by DzikuVx

padcom, 2022-02-12:
> Dear Cradylab,
> 
> I am flying a 690 size hexa and my quite good flying pids are:
> 
> Roll : P 155 I 35 D 45 Pitch: P 165 I 35 D 65 Yaw: P 145 I 55 D 0
> 
> I am currently on version 2.2.1
> 
> This should fly reasonable for a first try.
> 
> I hope you find this info useful. [image]

@Vissie26 It works for me as a charm so far (only flown a bit to the side and up and down on a pretty calm evening). My setup is as follows:

Tarot Ironman 690
Tarot 4006 620kV with 13" carbon props
INav 4.0.0 - every other setting left at stock besides flight modes and receiver configuration)

This has finally enabled me to switch from ardupilot to inav! Thank you!!!

CradyLab, 2022-02-12:
Dzięki Matthias!  I'll give that a try when the weather warms up in the spring.  Cheers!

pDR8iLQ4go, 2022-02-12:
Nice to see that others are still flying this build. Meanwhile I‘m more a fan of 3“ and 5“ freestyle copters. But my 12“ Tarot is still alive. And I‘m planning to test INAV 4.1. Maybe we can share our settings?

CradyLab, 2022-02-12:
If I successfully get my hex back in the air in the next couple of months I'll be sure to post my settings as well

padcom, 2022-02-12:
> Nice to see that others are still flying this build. Meanwhile I‘m more a fan of 3“ and 5“ freestyle copters. But my 12“ Tarot is still alive. And I‘m planning to test INAV 4.1. Maybe we can share our settings?

I think we should put together a site where settings are shared so that if you have doubts what motors and props to select for a given frame, you'll be able to see what has worked for other people. For example this was my 3rd attempt at INav in my hexa and had it not been for this thread I'd not be able to get it to work. I was basically going down with the values instead of up. I think if I'd have had some examples to draw conclusions from, I'd be on my way about a year sooner.

CradyLab, 2022-02-12:
> 

Excellent idea.  LMK if you go ahead with that I'll contribute what I can

padcom, 2022-02-14:
Just FYI I have registered inav-settings.org and started working on a parser for CLI dumps so that you can upload them, the site will distil them, and then you'll be able to download just the settings you'd like: PIDs, mixers, OSD (that's my favorite) and maybe others too.

The parser is now ~30% complete which puts us ~2 months from this site being fully operational. :)

padcom, 2022-02-14:
So - the parser is done. Next: backend to store your uploads. 

@CradyLab  Would you like to do some testing when it is available?

CradyLab, 2022-02-14:
> So - the parser is done. Next: backend to store your uploads.
> 
> @CradyLab Would you like to do some testing when it is available?

Of course!  Thanks for the invite =)

pDR8iLQ4go, 2022-02-19:
My 12“ build is flying with INAV 4.1.0. I‘m excited to see & compare the PID and filter settings with you.

CliffAndKajun, 2022-03-01:
I'm building a 15" x8 with the Matek 765-wing. Running Inav 4.1.. Would definitely be interested in sharing/learning pid and filters for something this size.

pDR8iLQ4go, 2022-03-31:
@padcom : Is there any progress concerning your parser?

padcom, 2022-04-01:
@pDR8iLQ4go There sure is - just going extremely slow :)

I started by re-doing the visual OSD editor in such a way that it is actually usable. That part is done. The next challenge that I have decided to undertake is to do it as a web application instead of a standalone Electon- or NW.js-based application. The same thing is happening for Bluejay and other hardware (both serial-based and bluetooth based) and I think it is much better than to have to download a new version each time.

Next thing that needs to happen is a library that will allow for communication between the controller and the application. That part is happening, although on a much slower pace. I want to make it at the very least functional to the extend where I can read and write configuration messages (especially the ones pertaining to OSD configuration).

Having both of those in place the next step is to put it all together, create a DB-based backend (I'm thinking Hassura on top of postgresql) to store the settings and a bit of user interface for selecting and then modifying the OSD layout.

After that part is done the next step is to incorporate reading of PID settings and storing/manipulating them the same way.

Hope that makes at least a little bit of sense.

petrweigl, 2024-03-03:
Are you continuing somehow? Tried inav 7 on big copter?

padcom, 2024-03-04:
As for inav-settings.org I still pay for the domain, but I am willing to donate it to the project if there is someone brave enough to make some use of it.

pDR8iLQ4go, 2024-03-04:
That‘s funny. Last weekend, I‘ve been flashing ELRS on the R9slim+ of my 12“. To be able to test it, I installed INAV 7. Next weekend I‘m able to report if INAV7 can be used on a 12“ hexacopter. Fingers crossed…

pDR8iLQ4go, 2024-03-10:
I successfully updated to INAV 7.0.0. Did not spend too much time on testing but my 12" Hexacopter is flying well with these PIDs:

```
set mc_p_pitch = 140
set mc_i_pitch = 45
set mc_d_pitch = 70
set mc_p_roll = 120
set mc_i_roll = 40
set mc_d_roll = 60
set mc_p_yaw = 180
set mc_i_yaw = 55
set mc_p_level = 15
set mc_i_level = 10
set dterm_lpf_hz = 30
```

[2024-03-10_diff.txt](https://github.com/iNavFlight/inav/files/14549793/2024-03-10_diff.txt)

petrweigl, 2024-03-10:
thank you for reference, please what values do you have on these? (feed forward slider in PID settings table)

set mc_cd_pitch = 50
set mc_cd_roll = 50
set mc_cd_yaw = 50

Antutank, 2025-02-25:
hi
can someone share the PID for 15" fpv

animesh-tech, 2025-09-21:
Hii I have  6 inch Deadcat fraem can someone give they PID dump pl