iNavFlight/inav #5260: Servos get stuck and "lazy" when my flying wing is on angle/horizon mode

freerider4real, 2019-12-08:
Hi;
I'm using inav for a flying wing, with fpv gear and gps. When I switch into angle or horizon the servos dont react as expected. Testing on the ground, they move as they should do in the air, however, when its actually flying, as soon as I switch modes it dives like crazy going out of control. By testing on the ground, I discovered that they like to stick in a certain position. For example, if I give pitch up with my stick, it tends to remain there, eventhough the airplane is flat. 
I've check alignmets, mixer, calibration... it could be maybe pids.... 
I dont know what to do next, please help!
I'm spanish, so I accept replies in spanish

freerider4real, 2019-12-08:
By the way, im using matek F405

digitalentity, 2019-12-08:
Please check this video [The most common iNav mistakes](https://youtu.be/fVJYodLimD8) - if it dives when you switch to angle mode - most likely FC alignment is not correct.

Servo "sticky" behavior on the ground is expected and was explained multiple times. When you center the stick it's the command to the FC to keep the airplane position. Because airplane keeps position with deflected servos FC doesn't re-center them.

freerider4real, 2019-12-08:
Thanks! Its true that my board is rotated 90 degress anti clockwise, but I have specified that in inav, in the configurator tab. Nevertheless I will try to rotate it to its natural position. (Hope I dont mess up wires, space is tight)

freerider4real, 2019-12-09:
I've just tried the new setup. Still doesnt work. It first rolls about 90 degress and the dives wile rolling

DzikuVx, 2019-12-24:
This looks like a setup issue. Did you reversed any channels on the radio? Did you correctly set board orinattion? Did you reversed one servo?
Are servos moving correctly in Many, Acro/Air and Angle mode?

closed 2019-12-24 by DzikuVx