iNavFlight/inav #5262: pid_type problem

hali9, 2019-12-08:
Since https://github.com/iNavFlight/inav/pull/5129 is CLI pid_type. I have set this to AUTO or PIFF, but I see in configurator (and blackbox) default PID for multirotor and I have set platform_type = AIRPLANE.
As I look in code the usedPidControllerType is set in pidInit() in activateConfig() in readEEPROM() in init().
ENABLE_STATE(FIXED_WING) is set in mixerUpdateStateFlags() also in init() but after readEEPROM()
So usedPidControllerType is always zero and 
`static inline const pidBank_t * pidBank(void) { return usedPidControllerType == PID_TYPE_PIFF ? &pidProfile()->bank_fw : &pidProfile()->bank_mc; }`
 always return bank_mc.

usedPidControllerType should be set up in init() after mixerUpdateStateFlags()

stronnag, 2019-12-08:
See #5261

DzikuVx, 2019-12-09:
yeap, fixed in #5261

closed 2019-12-09 by DzikuVx