iNavFlight/inav #5263: Can the maximum height of the surface mode be adjusted?

multinnov, 2019-12-09:
Hello,
When my drone is in surface with lidar it can't go higher than 80 cm but I want go at 2 meters, can I config height max of this mode ?

Materials:
- Kakute F7
- Matek Optic Flow et Lidar sensor 3901-L0X

whoim2, 2019-12-13:
Altitude - its laser sensor vl53l0x on matek module. Rangefinder.
https://github.com/iNavFlight/inav/blob/master/src/main/drivers/rangefinder/rangefinder_vl53l0x.c

#define VL53L0X_MAX_RANGE_CM

digitalentity, 2019-12-13:
If you have a sensor that works **reliably** up to 2m - sure, it's configurable via CLI.

whoim2, 2019-12-13:
> If you have a sensor that works **reliably** up to 2m - sure, it's configurable via CLI.

vl53l1x up to 4m, from description, but not supported in inav (

digitalentity, 2019-12-13:
I have been testing vl53l1x. Over a white sheet in dusk areas L1X indeed outperform the L0X and is reliable enough up to about 3m distance (compared to 1.8-2m of L0X). 

However over real-world surfaces (tiled flooring, moderately lit area) they both are good only up to 1.5-1.8m, but L1X also has a big drawback - it can't measure distance less than 5-10cm.

whoim2, 2019-12-13:
> I have been testing vl53l1x. Over a white sheet in dusk areas L1X indeed outperform the L0X and is reliable enough up to about 3m distance (compared to 1.8-2m of L0X).
>
> However over real-world [...]

L1X work as L0X, selected in inav configurator?
Maybe, use optical lens for l0x/l1x for best work?

multinnov, 2019-12-16:
> If you have a sensor that works **reliably** up to 2m - sure, it's configurable via CLI.

I thank you. But how do this in CLI? thank you

thijn1234, 2020-01-09:
its verry simple 
get inav_max_surface_altitude = 1000
inav_max_surface_altitude = 1000
Allowed range: 0 - 1000

for 2 m type
set nav_max_surface_altitude = 200

thijn1234, 2020-01-09:
when you type in cli
get alt you will see all alt settings

closed 2020-03-23 by stale[bot]