iNavFlight/inav #5265: SRXL2 or SRXLv2 Support

CloudsCapone, 2019-12-10:
## Current Behavior
<currently, I do not see available protocol to use the newer Spektrum SRXL2 receiver>

## Desired Behavior
<I would love to see option or support>

## Suggested Solution
<If somewhat easily achievable, would be happy with a simple string of CLI commands for the time being.>

## Who does this impact? Who is this for?
<I believe this impacts anyone wishing to use the receiver with iNav>

DzikuVx, 2019-12-10:
Hi,
So far Spectrum was not very eager to work with us to provide their radio protocols support in INAV. Without Spectrum assistance and probably hardware, INAV developers have zero to none chances to add support.
I suggest contacting Spectrum customer center with a request to support iNAV

SpektrumRC, 2019-12-11:
Unless we missed an email, I don't recall having been reached out to regarding this, other than some other posts on here from users requesting it.
Our SRXL2 protocol is public and available at https://github.com/SpektrumRC/SRXL2 including both documentation and C library files.
I had previously looked to see if it would have been quick enough to do a merge from betaflight srxl2 implementation into iNav, but there just seemed to be too many other changes/dependencies  that are not yet merged to iNav, and unfortunately I've been too busy to make a new implementation specific to iNav's current state.

donsinger1, 2019-12-27:
Dzikuvx and SpektrumRC...I asked about this very issue and need in the RC Groups thread on the new SRXL2 receivers, specifically the 6610T and 637T....

I want to use the SRXL2 port on the 6610T to connect to a wing aware flight controller, such as a Matek 411WSE for an iNav build.  I know it has been rolled out on Betaflight (4.1 and above), but wanted to know if there was a list of boards for iNav that had a target that included the SRXL2 protocol.

Andy Kunz replied that I should ask here and "bug the developers" to implement.

From your discussion, to the outsider, it looks like people are pointing fingers at each other, 'everyone is busy', so nothing gets done....It would be great if the development community would embrace Spektrum's new focus on openness and willing to cooperate with 3rd parties; and, I hope that Spektrum does not continue to be snarky as per the reply above and rather respond like 'well, never saw the email but let's work together to get this done'...

Can us users hope to see some progress on this issue? I do not want to have to purchase a FrSky system just to get this capability....

Thanks much for listening...Best, Don

MiguelFAlvarez, 2020-01-16:
We've still been pretty swamped with stuff at work, but I've taken some personal time to work through adding this. VERY EXPERIMENTAL! Testing/Feedback would be appreciated though.
@donsinger1 Not sure if a 6610T would work, as it handshakes slightly differently than the standalone serial receivers do.

https://github.com/MiguelFAlvarez/inav/releases/tag/2.3.0-SRXL2-TEST

donsinger1, 2020-01-16:
Hi Miguel:

You are a Prince!!!!

This would be fantastic if it worked on the 6610T…..I have one ready to test and a few blank Matek boards to try on…since I am doing a build right now, I can incorporate this with the FrSky RX with SBUS I had planned to use and test servos, esc, etc…will do the RX’s in parallel.

Can’t wait to play with this tonight.

Cordially,
Don Singer
KF7DS

https://github.com/MiguelFAlvarez/inav/releases/tag/2.3.0-SRXL2-TEST<https://eur04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FMiguelFAlvarez%2Finav%2Freleases%2Ftag%2F2.3.0-SRXL2-TEST&data=02%7C01%7C%7Cdc045a1c78e241b0f65608d79a42b50c%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637147483601412499&sdata=2YMI20Zy89r%2ByrEiOeBJRKvMrVIK%2FoqZFBaGMYsnehA%3D&reserved=0>

donsinger1, 2020-01-17:
Hi Miguel:

I also have a few 4650s so will try those as well.

Don

https://github.com/MiguelFAlvarez/inav/releases/tag/2.3.0-SRXL2-TEST<https://eur04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FMiguelFAlvarez%2Finav%2Freleases%2Ftag%2F2.3.0-SRXL2-TEST&data=02%7C01%7C%7Cdc045a1c78e241b0f65608d79a42b50c%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637147483601412499&sdata=2YMI20Zy89r%2ByrEiOeBJRKvMrVIK%2FoqZFBaGMYsnehA%3D&reserved=0>

Romulusg, 2020-02-04:
Has any of this been included in 2.4? Or do I need to use 2.3.0? Thanks for the work on this.

MiguelFAlvarez, 2020-02-04:
> Has any of this been included in 2.4? Or do I need to use 2.3.0? Thanks for the work on this.

I haven't received any feedback from anybody, and the only testing that has been done is from my own personal use (no issues so far), so I haven't made a merge request yet. As of right now, it's only available on 2.3 from the custom builds on my page. Haven't compiled for 2.4 yet since its still in the release candidate phase.

Romulusg, 2020-02-04:
Thanks for the info. I will test it next weekend when I can. What type of feedback would you like? 

Todd

MiguelFAlvarez, 2020-02-04:
@Romulusg Mainly just set it up and make sure it works properly (channel data, telemetry, rssi, failsafe detection) and doesn't cause any issues to otherwise normal operation like FC freezing/crashing. I only have F4 FC's so that's all I've tested personally. No idea if it works for F7's and F3's dont have enough flash to support the feature, so it's not implemented for those. Even if you test on F4's just be cautious and test everything on the bench. While it's not likely, its possible that some combination of features might cause an issue that I'm not seeing.

Romulusg, 2020-02-11:
I wasnt able to test this weekend I couldnt get access to my srxl2 receiver.  I ordered another one and will update as I test.

Romulusg, 2020-02-20:
Hi Miguel,

I have loaded the hex file and copied the cli info.  So far on my matekf411 wing FC everything appears working.  controls function as expected and modes switch as well.

Auto launch works on the bench with a prop off and auto level is working as well 

I havent tried RTH as it is on the bench.

So far it seems good.  

I will try a flight when the windy chinook wind in calgary drops hopefully this weekend.

Romulusg, 2020-02-20:
I installed the spm4650 spektrum rx and put it on TX1

one side note in the CLI the receiver is SRXL2 and it shows as valid but it doesnt appear in the receiver drop down.  My guess it the array for the list range needs to be expanded by a row as it appears last in the list.

SpektrumRC, 2020-02-20:
@Romulusg Thanks for the feedback. Look forward to hearing more.
In regards to the configurator receiver drop down, it wont appear there until the configurator is also updated. For now, the cli is the only option to set this. I will make the appropriate configurator changes as well once a merge request for the protocol is submitted.

Romulusg, 2020-02-21:
Ok additional testing this evening.  I was able to enter the betaflight menu and change vtx settings as expected. another thumbs up.

SpektrumRC, 2020-02-21:
@Romulusg btw, I compiled these changes onto the 2.4 release last night, but did not get a chance to test. Will try to test tonight and if it looks good on the bench, I'll upload that

trickapm, 2020-02-27:
I can test a 4650 receiver with a matek f722 if i can get a link to the 2.4.0 builds?

MiguelFAlvarez, 2020-02-27:
@trickapm https://github.com/MiguelFAlvarez/inav/releases

trickapm, 2020-03-02:
On the bench all looks well, didn't notice any issues. But i may not really know what to look for either ;-). I already bound it to my tx using the button on the pcb so didn't use the bind_rx command.

AntDX316, 2020-03-11:
need SRXL2 support on 2.4.1

Romulusg, 2020-03-11:
@AntDX316 above there is a link to 2.4.0 there has not been an merge yet as far as @MiguelFAlvarez has mentioned in this page.

AntDX316, 2020-03-11:
> @AntDX316 above there is a link to 2.4.0 there has not been an merge yet as far as @MiguelFAlvarez has mentioned in this page.

Do I have to reflash to the firmwares underneath the post or can I just drop the CLI in 2.4.1??

Romulusg, 2020-03-11:
@MiguelFAlvarez can correct me if I’m incorrect but I believe the code needed for the flight controller hex files is in the link above. You would need to flash to that version as the cli commands are reliant on the code in the background.

AntDX316, 2020-03-11:
> @MiguelFAlvarez can correct me if I’m incorrect but I believe the code needed for the flight controller hex files is in the link above. You would need to flash to that version as the cli commands are reliant on the code in the background.

There are a lot of files that are in the hex.. I'm sure it wasn't done individually?  It much be an easy process to inject?  Now, can we get the steps required to inject the code for it to work?  Just incase the maker stops supporting for some reason?

Romulusg, 2020-03-26:
@MiguelFAlvarez i flew my inav setup wing with the srxl2 receiver and everything worked seamlessly. How does this get merged?

donsinger1, 2020-03-26:
That is great…been way too busy the last three months to contribute, but glad to see there is substantial progress. Thanks Miguel and everyone else.

Stay safe.

Best
Don Singer
Portland, OR

MiguelFAlvarez, 2020-03-26:
I'll start doing pull requests for the necessary changes

MiguelFAlvarez, 2020-03-26:
First pull request for uart idle callback support https://github.com/iNavFlight/inav/pull/5540 which is required for SRXL2 to work.
Once that's merged, I will make a pull request for the SRXL2 support.

AntDX316, 2020-03-26:
> First pull request for uart idle callback support #5540 which is required for SRXL2 to work.
> Once that's merged, I will make a pull request for the SRXL2 support.

Are you going to update it to 2.4.1?

MiguelFAlvarez, 2020-03-26:
> > First pull request for uart idle callback support #5540 which is required for SRXL2 to work.
> > Once that's merged, I will make a pull request for the SRXL2 support.
> 
> Are you going to update it to 2.4.1?

I'm making a pull request for the master branch. From there it should start getting included in upcoming releases. As for including it in my personal builds, I've only been compiling my changes on the latest release, and there is no 2.4.1 release yet unless I'm missing something, only 2.4.0. https://github.com/iNavFlight/inav/releases

AntDX316, 2020-03-27:
> I'm making a pull request for the master branch. From there it should start getting included in upcoming releases. As for including it in my personal builds, I've only been compiling my changes on [...]

What do you mean a pull request from the master branch?  How long could that possibly take and is it even worth upgrading to 2.4.1 from 2.4.0? : )

MiguelFAlvarez, 2020-03-27:
@AntDX316 I'm not sure how long it'll take. It's up to the inav devs when they have time to review/approve the code changes, and when they plan on making the next official release.
I'm not sure where you're getting 2.4.1. There is no 2.4.1 inav release yet.

AntDX316, 2020-03-27:
> @AntDX316 I'm not sure how long it'll take. It's up to the inav devs when they have time to review/approve the code changes, and when they plan on making the next official release.
> I'm not sure where you're getting 2.4.1. There is no 2.4.1 inav release yet.

2.4.1 has been out
https://github.com/iNavFlight/inav-configurator/releases

MiguelFAlvarez, 2020-03-27:
That is the configurator, not the firmware.

AntDX316, 2020-03-27:
> That is the configurator, not the firmware.

so 2.4.0 that you put the SRXL2 on is the latest? : )

MiguelFAlvarez, 2020-03-27:
> > That is the configurator, not the firmware.
> 
> so 2.4.0 that you put the SRXL2 on is the latest? : )

yup!

AntDX316, 2020-03-28:
> > > That is the configurator, not the firmware.
> > 
> > 
> > so 2.4.0 that you put the SRXL2 on is the latest? : )
> 
> yup!

I'm trying to flash with load firmware [local] w/ the fix but it doesn't work? : (

AntDX316, 2020-03-28:
> > > That is the configurator, not the firmware.
> > 
> > 
> > so 2.4.0 that you put the SRXL2 on is the latest? : )
> 
> yup!

Can you push it through so we can have SRXL2 native support with 2.5 so I can load firmware [Online] and it work?

AntDX316, 2020-04-03:
> > > That is the configurator, not the firmware.
> > 
> > 
> > so 2.4.0 that you put the SRXL2 on is the latest? : )
> 
> yup!

What do you set in receiver protocol? 2048??

AntDX316, 2020-04-03:
> > > That is the configurator, not the firmware.
> > 
> > 
> > so 2.4.0 that you put the SRXL2 on is the latest? : )
> 
> yup!

Works perfectly!

Romulusg, 2020-04-29:
With the release of 2.5 has the merge of srxl2 been added?  I dont see it in the release notes. @MiguelFAlvarez

MiguelFAlvarez, 2020-04-29:
Not yet, the devs requested some changes to the code that I've just not had a chance to address

AntDX316, 2020-04-30:
> Not yet, the devs requested some changes to the code that I've just not had a chance to address

Please have it where there is native SRXL2 support w/ no workaround.

Romulusg, 2020-04-30:
Ok thanks for the update.

rickgode, 2020-05-18:
Hi guys!  I flashed the hex above to my matek 405 mini and using a SPM4650 Rx and it seems to work perfectly on the bench.

donsinger1, 2020-05-18:
What version…2.4 or 2.5?

Don

rickgode, 2020-05-18:
Mine says

# version
# INAV/MATEKF405 2.3.0 Jan 15 2020 / 22:30:44 (e5b5aad2c)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Ricky

tourerjim, 2020-05-26:
For Inav 2.5 could someone guide for the SRXL2 file so I can use my Sppedybee F4 with the SPM4650 receiver.

AntDX316, 2020-05-26:
What takes so long to get native SRXL2 support?  Betaflight already has it.

tourerjim, 2020-05-29:
> For Inav 2.5 could someone guide for the SRXL2 file so I can use my Sppedybee F4 with the SPM4650 receiver.

I've now got my quad built and have Inav 2.5 loaded on the Speedybee FC but without the SRXL2 I can't complete it and just having retired would loved to get out and try it, I copped the above Cli SRXL2 file but doesn't work, Help

MiguelFAlvarez, 2020-05-29:
@tourerjim inav doesn't officially support it yet. I'm working on that. You need a custom build from here in the meantime https://github.com/MiguelFAlvarez/inav/releases

Romulusg, 2020-05-29:
> 
> 
> @tourerjim inav doesn't officially support it yet. I'm working on that. You need a custom build from here in the meantime https://github.com/MiguelFAlvarez/inav/releases

@MiguelFAlvarez does that link and hex files then work with version 2.5? 

Thanks

MiguelFAlvarez, 2020-05-29:
In case anyone is keeping track of the state of implementation, PR #5757 has been approved (uart driver dependencies for SRXL2). Now the actual SRXL2 merge request has been submitted as PR #5791

rickgode, 2020-05-29:
Sweet!  I was able to maiden my 2.4 inav with SRXL2 support for the SPM4650 Rx on a Matek 405 mini and it works great.  rssi works along with the telemetry TEXT GEN.

Here is some flight cam footage https://youtu.be/APn3u8ktu9E

Texting out auto launch https://youtu.be/ftnLqZ7Sk5c

tourerjim, 2020-05-30:
I've paste the SRXL2 file into Cli but get these Errors anyone know what could I be doing wrong.
[image]

MiguelFAlvarez, 2020-05-30:
> I've paste the SRXL2 file into Cli but get these Errors anyone know what could I be doing wrong.
> [image]

You need to install the custom firmware I linked above. That's why it thinks those parameters are invalid

tourerjim, 2020-05-31:
> @tourerjim inav doesn't officially support it yet. I'm working on that. You need a custom build from here in the meantime https://github.com/MiguelFAlvarez/inav/releases

Really appreciate your help, now got my SPM4650 working with Inav so now going to go through and set everything up,Thanks

tourerjim, 2020-05-31:
> > @tourerjim inav doesn't officially support it yet. I'm working on that. You need a custom build from here in the meantime https://github.com/MiguelFAlvarez/inav/releases
> 
> @MiguelFAlvarez does that link and hex files then work with version 2.5?
> 
> Thanks

Yes, I've got mine working

AntDX316, 2020-05-31:
> In case anyone is keeping track of the state of implementation, PR #5757 has been approved (uart driver dependencies for SRXL2). Now the actual SRXL2 merge request has been submitted as PR #5791

let us know when it's fully implemented so we can download it w/o a workaround : )

donsinger1, 2020-06-01:
Thanks much Miguel…been patient and appreciate your work on this…

Don

[30 more replies]

closed 2020-11-26 by DzikuVx