iNavFlight/inav #5269: Will a Luminier F-7 LUX Fc w/ INAV work with US-100 Sonar and if it can is it best to use m5 and m6 pins?

rhoads6257, 2019-12-13:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)
* [How to setup INAV on a flying wing](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLn_lCX7UB2OPLzR4w5G4el3)
* [How to setup INAV on a 5" quad](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLl9M6Vm-ZNLMCr6_9yNVHQG)

DzikuVx, 2019-12-14:
US-100 in trigger/echo mode is not supported by any new hardware

closed 2019-12-14 by DzikuVx

rhoads6257, 2019-12-14:
What sonar does work with new hardware? Is sonar just irrelevant at this
point!

DzikuVx, 2019-12-16:
Sonars are rather useless on multirotors and even when they work, range and accuracy is very reduced. 
If you want, you can interface US100 with INAV using this DIY board: https://github.com/iNavFlight/inav-rangefinder

rhoads6257, 2019-12-16:
Thanks! I think I'll skip sonar.