iNavFlight/inav #5270: Pit Mode via Global Function

swissfreek, 2019-12-14:
## Current Behavior
Currently Power Level can be changed via Global Function, and for SmartAudio it can be set from 0-3. For a 800mW transmitter, this corresponds to 25mw, 200mW, 500mW, 800mW. There is no way to engage any power level lower than 25mW (aka Pit Mode).

## Desired Behavior
It would be nice if it were possible to enable and disable Pit Mode so that you can be in Pit Mode until you flip a switch or something.

## Suggested Solution
Add a SET_PIT_MODE to the Global Functions so that it can be controlled via Logic Condition similar to SET_VTX_POWER.

## Who does this impact? Who is this for?
Anyone who flies with TBS Unify VTXs and wishes to control Pit Mode, especially those who fly in groups where they don't want to swamp neighbors until they are flying.

padcom, 2021-01-26:
Seriously, this should be simple enough. Just add a mode that will enable pit. It is already there in betaflight.

sixtyfive, 2023-02-05:
6.0.0 is about to be released. Has this Issue received any consideration by the developers?

OptimusTi, 2023-02-06:
This is a request. Not an issue.

sixtyfive, 2023-02-06:
Sorry, OptimusTi, by Issue (uppercase I) I merely meant to refer to the thing that GitHub decided to call so. Not the lowercase-i issue, as in the thing otherwise known as a problem or similar.

closed 2024-02-02 by DzikuVx