iNavFlight/inav #5278: More than 2 motors on an airplane with differential thrust

alex3s, 2019-12-19:
What is the method to use more than 2 motors on a flying wing?
I have 4 counterrotating motors, similar to Parrot Swing, but INAV configurator 2.3.2 allows only 2 motors, 
the rest is always servos, no matter what I do.

The CLI command below doesn't help, as the output mapping never shows motor3 (picture)
mmix reset
mmix 0 1.0 0.0 0.0 0.3 # Left Engine
mmix 1 1.0 0.0 0.0 -0.3 # Right Engine
mmix 2 1.0 0.0 1.0 0.0 # Back Engine

Configurator screen:
Output | S1 | S2 | S3 | S4 | S5 | S6 | S7 | S8 | S9 | S10
Motor 1 | Motor 2 | - | - | - | - | - | - | - | -

How can I add motor3 or motor4, either in source files or configurator or CLI?

DzikuVx, 2020-01-01:
Arirplane mixer accepts only 2 motors. You can use a ESC connected to servo output

closed 2020-01-01 by DzikuVx

alex3s, 2020-01-01:
Hi Pawel,
Thank you for your answer. I'm trying to construct an airplane with only motors, using an Spracing Brushed FC, this one: https://www.aliexpress.com/item/32734941363.html
It has 6 motor outputs but no servo output.
When installed INAV (different versions, including the latest) only 2 of them ever work when the FC is in AIRPLANE configuration.
I wonder which part of the source code has the limit, so I can change it myself and use all 6 motors to stabilise the airplane in all 6 axes.
I've tried to use it in TRICOPTER configuration with some success - the 3rd motor works, but it doesn't in AIRPLANE configuration - when I set the 3rd motor as being the servo, the motor connected to that output buzzes, but doesn't spin.
I can compile it myself on Linux, but can't find the correct file to modify the max number of motors for an airplane. Is it a difficult modification?