iNavFlight/inav #5281: how do I install the us-100 sonar

thijn1234, 2019-12-20:
Hi 
I have a kakutef7mini FC and I have already set a lidar on my drone but I aad the us-100 and remove the lidar but I can't connect the sonar he turns red . can someone help me?

# version
# INAV/KAKUTEF7MINI 2.3.0 Nov 24 2019 / 10:31:13 (d0bdd38c0)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# gf

# feature
feature THR_VBAT_COMP
feature SOFTSERIAL
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ON_USB
beeper -LAUNCH_MODE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# map
map TAER

# serial
serial 1 65536 115200 9600 0 115200

# aux
aux 0 0 0 1700 2100
aux 1 1 0 1700 2100
aux 2 26 0 1700 2100
aux 3 3 1 1600 2100
aux 4 24 1 1775 2100
aux 5 11 4 1700 2100
aux 6 18 5 1700 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 23 0 H
osd_layout 0 1 12 1 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 12 11 V
osd_layout 0 9 1 2 H
osd_layout 0 11 1 3 H
osd_layout 0 12 0 4 H
osd_layout 0 15 2 0 V
osd_layout 0 28 22 1 V
osd_layout 0 30 0 10 V
osd_layout 0 33 0 1 V
osd_layout 0 86 1 11 V

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 150
set gyro_stage2_lowpass_hz = 150
set acc_hardware = ICM20689
set acczero_x = 12
set acczero_y = -12
set acczero_z = -39
set accgain_x = 4093
set accgain_y = 4092
set accgain_z = 4040
set rangefinder_hardware = HCSR04
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set min_throttle = 1050
set max_throttle = 2000
set motor_pwm_rate = 490
set failsafe_throttle = 1385
set align_board_roll = 2
set model_preview_type = 3
set small_angle = 10
set applied_defaults = 1
set alt_hold_deadband = 10
set mc_airmode_threshold = 1000
set inav_use_gps_velned = OFF
set inav_max_surface_altitude = 1000
set inav_w_z_baro_p =  0.500
set inav_w_z_gps_p =  0.100
set inav_w_z_gps_v =  0.500
set nav_use_midthr_for_althold = ON
set nav_user_control_mode = CRUISE
set nav_auto_speed = 10
set nav_auto_climb_rate = 30
set nav_manual_speed = 10
set nav_manual_climb_rate = 100
set nav_land_slowdown_maxalt = 1200
set nav_emerg_landing_speed = 100
set nav_max_terrain_follow_alt = 1000
set osd_alt_alarm = 11
set osd_dist_alarm = 200
set osd_neg_alt_alarm = 1

# profile
profile 1

set max_angle_inclination_rll = 150
set max_angle_inclination_pit = 150
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set heading_hold_rate_limit = 60
set nav_mc_pos_z_p = 5
set nav_mc_vel_z_p = 50
set nav_mc_vel_z_i = 28
set nav_mc_vel_z_d = 8
set nav_mc_pos_xy_p = 80
set nav_mc_vel_xy_p = 50
set nav_mc_vel_xy_i = 40
set nav_mc_vel_xy_d = 60
set mc_iterm_relax = RPY
set mc_iterm_relax_cutoff = 20
set d_boost_factor =  1.000
set thr_mid = 49
set thr_expo = 100
set rc_expo = 100
set rc_yaw_expo = 70
set roll_rate = 6
set pitch_rate = 6
set yaw_rate = 5
set manual_rc_expo = 100
set manual_rc_yaw_expo = 70

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 320

# end the command batch
batch end

DzikuVx, 2019-12-24:
US-100 is not supported directly. It can be connected using: https://github.com/iNavFlight/inav-rangefinder

closed 2019-12-24 by DzikuVx

thijn1234, 2020-01-02:
@DzikuVx  I test it with an arudio uno between the us-100 and the kakute f7 mini 
but it the sonnar is stil red 
#define DEBUG
#include <SoftwareSerial.h>;
 
#define US100_TX 3
#define US100_RX 2
#define LED_PIN 13

SoftwareSerial US100(US100_RX, US100_TX);

#define I2C_SLAVE_ADDRESS 0x14

#define STATUS_OK 0
#define STATUS_OUT_OF_RANGE 1

#include <Wire.h>

#ifndef TWI_RX_BUFFER_SIZE
  #define TWI_RX_BUFFER_SIZE ( 16 )
#endif

uint8_t i2c_regs[] =
{
    0, //status
    0, //older 8 of distance
    0, //younger 8 of distance
};

const byte reg_size = sizeof(i2c_regs);
volatile byte reg_position = 0;

void requestEvent()
{  
    Wire.write(i2c_regs, 3);
}

void receiveEvent(uint8_t howMany) {

    if (howMany < 1 || howMany > TWI_RX_BUFFER_SIZE) {
        // Sanity-check
        return;
    }
    reg_position = Wire.read();

    howMany--;

    if (!howMany) {
        // This write was only to set the buffer for next read
        return;
    }

    // Everything above 1 byte is something we do not care, so just get it from bus as send to /dev/null
    while(howMany--) {
      Wire.read();
    }
}

void setup() {

#ifdef DEBUG
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
#endif

  pinMode(US100_RX, INPUT);
  pinMode(US100_TX, OUTPUT);
  pinMode(LED_PIN, OUTPUT);
[... 1149 lines cut ...]
set roll_rate = 6
set pitch_rate = 6
set yaw_rate = 5
set manual_rc_expo = 100
set manual_rc_yaw_expo = 70
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 320
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end