iNavFlight/inav #5282: Matek F411 wse - Softserial 2 -Mavlink Telemetry

pierpaolobirdi, 2019-12-21:
Hi

I can't get the telemetry in MAVLINK mode through the Softserial 2 port connected to TX2 of the F411 WSE.

If I configure TX2 as UART2 and I assign Telemetry of the type MAVLINK, it communicates perfectly and I read Telemetry perfectly with QGroundcontrol, but this UART is reserved for the Serial port of the signal receiver, so it must be done through softserial 2 that comes out of the TX2 pin of the FC.

When configuring Softserial 2 as telemetry of the MAVLINK Mode, nothing reaches Qgroundcontrol.

I have tried to set the parameters:
 
telemetry_uart_unidir = ON
telemetry_Invert = ON

But I have not been able to read telemetry.

pitersonfascist, 2020-01-05:
I faced the same issue. On Softserial 2 only LTM and SmartPort are working...
I've tried to re-map pins to have two-directional Softserial and connected GPS to it and used Serial 1 for the Mavlink, but got too many errors on Softserial so GPS became unstable.

I would also appreciate any solution to have stable Mavlink telemetry at least on 9600 speed

Urko1976, 2020-02-10:
Dear All,
I have the same issue with this F411wse as I cannot get telemetry. I am using a R9mm receptor and I could not make work Fport in the FC neither the sbus + smart port. Any suggestion?
Could it be that our FCs are part of a malfunction batch?
Best regards,
Urji

closed 2020-04-25 by stale[bot]