iNavFlight/inav #5283: Pitch/Roll/Yaw FF adjustment & PIFF in OSD

nikram77, 2019-12-21:
## Current Behavior
There is no "adjust FF" in adjustments for each axis.
There is no element for showing PIFF in OSD (only PID).
When PID is in osd, D shows FF value, but can`t be changed by D adjustment.

## Desired Behavior
Adjust FF function for each axis in adjustments (same as P&I&D) and PIFF element in OSD to show values (so you can see what adjustment is active and how value is changed).

## Suggested Solution
Have no idea how to code, so forgive me, but I can`t give any solution :)

## Who does this impact? Who is this for?
It is for fixed wings. Will allow to fine tune PIFF controller in-flight.

OlivierC-FR, 2019-12-23:
[image]

It's in the menu in 2.3, what version are you running ?

closed 2019-12-24 by DzikuVx

nikram77, 2019-12-24:
I got the same menu, but it is not what I`m talking about. I`m talking about in-flight adjustments .
You set them in Adjustment tab in configurator. IN below pic you see how I`ve got setup mine. P, I and D adjustment for roll and pitch. What I want to have is option of "Pitch/Roll FF Adjustment" same as for P, I, and D. P and I work perfectly, but I would like to do the same for FF. D adjustment in this setup doesn`t do anything. On the second pic you see OSD layout I`ve got for this "tune mode" showing PID`s. This shows me which value is adjusted (it blinks when activated and value changes) and how much. Currently it shows PID, selecting D adjustment  doesn`t change the value (although it blinks). Value shown for D is FF value. I hope I explained this clearly.
It` wasn`t a bug, it`s suposed to be Feature Request but bot decided differently...

[image]
[image]

P.S. Merry Christmas to everyone :)