iNavFlight/inav #5290: Motor remaining at idle with Dshot - Emergency Landing

Jetrell, 2019-12-23:
When using Dshot the motor will remain idling at low rpm's in a fixed wing, when the `emergency landing procedure` takes place.      e.g. `poor GPS` or `sanity check`.

At present using Dshot, the motor will not stop because Dshot auto calibrates the minimum throttle value.
While PWM protocols under auto throttle, will stop the motor if the `emergency landing procedure` is initiated.

Jetrell, 2019-12-30:
Has anyone else experienced this?
I only started using Dshot on planes with the release of 2.3.

The way to do a simple test is to turn off your radio when the FC has no GPS fix (indoors). The FC will go into failsafe and commence the  `emergency landing` procedure. 
In my case the motor still ticks over at 1000rpm. While it didn't with PWM, oneshot or multishot.

DerRCFan, 2021-01-29:
Confirm! 

BUT, as well with oneshot and multishot for me. INAV Version 2.6
While being in emergency landing procedure and setting up failsave_throttle = 1000, the wing remains at throttle_idle instead of cutting the throttle.
I have to set throttle_idle to zero to solve this problem (possible since V2.6)

closed 2023-12-28 by Jetrell