iNavFlight/inav #5291: Crossfire telemetry and custom firmware v.2.3.0

romko1, 2019-12-24:
Hello,
Santa brought me the Christmas TBS Micro RX and Nano TX for both of my drones.
In old FW 2.2.1 for MATEKF405 and MATEKF722mini I see only 5 or 6 telemetry data in Taranis X9D + (CRSF connected to UART). Before I used smartport, I saw more than 10 parameters in Taranis.
I have modified FW for TARGET - MATEK405: target.c and target.h and for TARGET - MATEKF722: target.h.
However, I wanted to create a new custom FW v.2.3.0, but when Linking MATEKF405 crashes the compilation.
Please someone to compile my own FW for the MATEKF405 and MATEKF722mini board and activate the CSFR telemetry for Taranis.
_Modification for MatekF722mini target.h: (mod line 182)_
#define AIRSPEED_ADC_CHANNEL           ADC_CHN_4

// *************** PINIO ***************************
#define USE_PINIO
#define USE_PINIOBOX
#define PINIO1_PIN                  PC8 // VTX power switcher
#define PINIO2_PIN                  **PB3 // Camera on/off**

// *************** LEDSTRIP ************************
#define USE_LED_STRIP

****************************************************************************************************
_Modification for MatekF405 target.c: (mod line 36)_
#endif
    DEF_TIM(TIM8, CH2, PC7,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 1), // S2  UP(2,1)
    DEF_TIM(TIM8, CH3, PC8,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 1), // S3  UP(2,1)
    DEF_TIM(TIM8, CH4, PC9,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 0), // S4  UP(2,1)

    DEF_TIM(TIM2, CH1, PA15, TIM_USE_PWM                                           , 0, 0), // S5  UP(1,7) - smartaudio VTX

#ifdef MATEKF405_SERVOS6

_Modification for MatekF405 target.h (mod line 122)_
#define USE_SOFTSERIAL2
#define SOFTSERIAL_2_TX_PIN      **PA15  //smartaudio VTX - S5**
#define SOFTSERIAL_2_RX_PIN      NONE  //free

Thank you.
R

romko1, 2019-12-27:
Hello,

I followed this procedure: 
https://github.com/iNavFlight/inav/blob/master/docs/development/Generic_Linux_development.md
and I still have a problem with Linking MATEKF405 :(

Please help. Thank you.
R

stronnag, 2019-12-27:
If you actually posted the compiler and  linker output then maybe someone can help you. Just saying it doesn't work is pointless.

romko1, 2019-12-27:
Sorry, here are 2 print-screens
[image]
[image]

stronnag, 2019-12-27:
Don't use a native Windows environment, it can't cope with a 32bit cross compiler in a 64bit run time.
Use Linux,  WSL or a Linux VM.

closed 2019-12-27 by stronnag

romko1, 2019-12-29:
@stronnag thank you for the guidance, but I do not currently have linux available. Can you generate the hex files, please? Thank you.

datamann, 2019-12-30:
Everyone with W10 can have Linux available. Just follow this easy to follow instructions.
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20with%20Linux%20Subsystem.md

romko1, 2020-01-01:
@datamann thank you, according to this procedure I created new firmware 2.3.0, but I had two warnings there. is it something important or is it OK?
Thank you for your response.
[image]