iNavFlight/inav #5293: Anyone willing to test my firmware target for Heli-Nation Talon F7 Fusion?

ricardodeazambuja, 2019-12-25:
Hi,

I managed to compile and make iNAV configurator recognise my flight controller Heli-Nation Talon F7 Fusion. On iNAV configurator it seems to be working. Here is my branch (the target name is CLRACINGF7 to keep the same name as from Betaflight):
https://github.com/ricardodeazambuja/inav/tree/talonf7fusion

I compiled it using the docker recipe from iNAV:
```
docker run --rm -v "$(pwd)":/home/src/ inav-build make TARGET=CLRACINGF7
``` 
Converted it to raw bin using (from inside the obj directory):
```
docker run --rm -v "$(pwd)":/home/src inav-build arm-none-eabi-objcopy -I ihex -O binary inav_2.3.0_CLRACINGF7.hex inav_2.3.0_CLRACINGF7.bin
``` 
and finally downloaded it to the flight controller using dfu-util (after starting the flight controller holding the DFU enabling button):
```
dfu-util -a 0 -s 0x08000000:leave -D inav_2.3.0_CLRACINGF7.bin
```
If you want to backup your original firmware, here is the command:
```
dfu-util -a 0 -U talonf7fusion_betaflight.bin -s 0x08000000:131072
```

Any help will be much appreciated ;)

**DISCLAIMER: I haven't flown anything using it, so totally untested!**

Cheers,

Ricardo

ricardodeazambuja, 2019-12-28:
Ok, I learned how to fix the order of the motors and everything seems to work. Tested the motors, armed and disarmed on the bench, that kind of stuff. Now it "only" needs to fly to confirm that it really works :)

BTW, using the inav-configurator that installs as a chrome app (instead of running from the repo using "npm install"  and "npm start .") allowed me to download the firmware to the flight controller directly from the generated .hex using inav-configurator firmware flasher and "Load Firmware (Local)", hurray!

closed 2020-03-12 by stale[bot]

vizariy666, 2020-11-25:
> Привет,
> 
> Мне удалось скомпилировать и заставить конфигуратор iNAV распознать мой контроллер полета Heli-Nation Talon F7 Fusion. В конфигураторе iNAV он, кажется, работает. Вот моя ветвь (целевое имя-CLRACINGF7, чтобы сохранить то же имя, что и у Betaflight):
> https://github.com/ricardodeazambuja/inav/tree/talonf7fusion
> 
> Я скомпилировал его, используя рецепт докера из iNAV:
> 
> ```
> docker run --rm -v "$(pwd)":/home/src/ inav-build make TARGET=CLRACINGF7
> ```
> 
> Преобразовал его в raw bin с помощью (из каталога obj):
> 
> ```
> docker run --rm -v "$(pwd)":/home/src inav-build arm-none-eabi-objcopy -I ihex -O binary inav_2.3.0_CLRACINGF7.hex inav_2.3.0_CLRACINGF7.bin
> ```
> 
> и, наконец, загрузил его в контроллер полета с помощью dfu-util (после запуска контроллера полета удерживая кнопку включения DFU):
> 
> ```
> dfu-util -a 0 -s 0x08000000:leave -D inav_2.3.0_CLRACINGF7.bin
> ```
> 
> If you want to backup your original firmware, here is the command:
> 
> ```
> dfu-util -a 0 -U talonf7fusion_betaflight.bin -s 0x08000000:131072
> ```
> 
> Any help will be much appreciated ;)
> 
> **DISCLAIMER: I haven't flown anything using it, so totally untested!**
> 
> Cheers,
> 
> Ricardo

Hello, I really want to install on talon f7 v1 inav, I do not understand anything about it, please share the finished firmware if it is available. thank you very much.

ricardodeazambuja, 2020-11-25:
Hi @vizariy666,
 
After I published the message above, I actually used the target I created with one of my drones and it worked, but still it's a risk because I haven't fully tested for other possible problems / side effects.

Here is the target (just copy this directory inside the "target" directory where you cloned the inav repo):
https://github.com/thecognifly/inav/tree/talonf7fusion/src/main/target/CLRACINGF7

And the instructions to compile the firmware:
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Docker.md

After that, you can use the inav-configurator to load it in your flight controller (my instructions using dfu-util were to make a backup copy of the original firmware, but I think you could restore it using betaflight-configurator instead):
https://github.com/iNavFlight/inav-configurator/releases

I hope this helps. If you really need me to compile the firmware for you, I would suggest you should not use inav with the talon f7 because bad things may happen when you use untested code and you can't fix the problems that will certainly appear by yourself.

Cheers,

Ricardo