iNavFlight/inav #5297: MATEK F405 CTR LED Strip flicker/random colors

Kalveo, 2019-12-29:
## Current Behavior

When trying to add a LED strip to Matek f405 CTR the LEDs turn the configured color for a very short time (<0.1s) then change to some random color. This continues indefinately.
Example: (Configured as solid blue) Blue -> (random) -> Blue -> (random) -> etc.

## Steps to Reproduce

1. Add LED strip to (5V, G, S5)
2. Enable LED strip in configurator
3. Configure LED(s) as a solid color

## Expected behavior

The configured LED(s) should have the configured color

## Suggested solution(s)

## Additional context

[PasteBin CLI Dump](https://pastebin.com/KWGNFYzK)

---

Hardware
*  MatekSystems F405-CTR

version
*  INAV/MATEKF405 2.3.0 Nov 24 2019 / 10:52:41 (d0bdd38c0)
*  GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

elepete, 2019-12-30:
Most leds are sensitive to overvoltage. Add a diode (e.g. 1N400x as 1A capable) in the 5V (VDD line) that drops the voltage by ~0.6V and the flicker should be gone.
Please reply if that fixes.

Kalveo, 2019-12-31:
This did in fact solve the problem. Thank you very much,

closed 2019-12-31 by Kalveo