iNavFlight/inav #5299: Mamba F405

ssombra, 2019-12-30:
Hi. I flashed INAV to my Mamba F405 stack and everything works as expected (barometer, GPS, compass, r9mm) except ESCs, specifically I can not get Dshot (none of them) working on motor 3, it starts spinning (more like glitching) without stick input as soon as I choose Dshot. Other motors are ok. It seems out of sync. When I try multishot and sync motors everything seems to work ok (haven’t flown yet but seems ok). I saw in RCGroups other person with a similar problem:

https://www.rcgroups.com/forums/showpost.php?p=43328343&postcount=184

Stav91, 2020-02-03:
No update on this?Seems to be a major bug for Mamba F405 MK2. I have had the same problem along with some other pilots in my community.In resources motor outputs are configured correctly...

digitalentity, 2020-02-03:
Is this issue still present on 2.4-RC2?

Stav91, 2020-02-04:
Yes issue is present on 2.4-RC2

digitalentity, 2020-02-08:
Should be fixed in https://github.com/iNavFlight/inav/pull/5394

closed 2020-02-08 by digitalentity