iNavFlight/inav #5301: RSSI states 0 when using F.port on Matek F722SE

TIMO88D, 2019-12-31:
## Current Behavior
When using a R-XSR with F.port firmware connected to a Matek F722SE running latest INAV firmware on UART 2 TX pad(or any other UART) the RSSI is not picked up by INAV. It always reads 0. I am able to get full receiver inputs in the receiver page and full telemetry is sent pack to the Taranis x9d+ (including RSSI, battery volt, GPS settings and mag). LUA scripts are working correctly.

## Steps to Reproduce
I have UART 2 set to serial port and Serial based receiver- Fport selected in config with telemetry button set on. In the Receiver tab I have disabled RSSI channel. I have re-discovered sensor settings in the Taranis (running in channel 1-8 mode with telemetry) and rebound multiple times. Have also tried different combinations of the following settings
set serialrx_inverted
set serialrx_halfduplex
set telemetry_inverted
set telemetry_uart_unidir
I do not have any of the the ports tab set to smart port or Frsky and have followed the oscar scarling insructions and Bardwell video multiple times (as well as others).

## Expected behavior
RSSI value should atualtically show in setup tab and in OSD (with RSSI enabled in OSD)

## Suggested solution(s)
It appears that INAV do not know to look at the Fport input to get RSSI info, Could this be related to the RSSI ADC channel command to tell it where to look? Or perhaps there is a scaling issue and that the signal reads less (for example from -100 to 0) and Inav will only show a positive value.

## Additional context
I flashed betaflight with the same hardware/pin selection and was able to get the RSSI working using the instructions.

---

- FC Board name and vendor: Matek F722 SE
- INAV version string: 2.3

simtcrom, 2020-01-04:
Disable analog rssi (adc), and disable rssi channel in receiver tab.
Run below commands in configurator cli tab,

**_set rssi_source = PROTOCOL
set rssi_max = 50_**

That should display rssi.

TIMO88D, 2020-01-05:
Cheers, this works with:
set rssi_source = PROTOCOL
set rssi_max = 50 set
serialrx_inverted=on
set serialrx_halfduplex=on 

Only issue is that if you lose reception completely or the TX turns off the rssi value stays as the last value it had. Anyway to make this go to 0 in this situation?

JDQuadRacer, 2020-01-31:
I have the same issue but with a Matek F411 wing FC and using an R9 slim+. The biggest Dilemma is that I have 2 other planes set up the same way and they are working fine... What can i try ????

TIMO88D, 2020-02-01:
If using  a F4 board try changing 
serialrx_inverted=off

JDQuadRacer, 2020-02-01:
On the Matek F411 board im using the cli commands are: inverted = ON   /   halfduplex = ON  and im running F-PORT. since my post i have figured it out, I'm runnung inav 2.3.3. and it has an issue with cutting the RSSI in half or registering "0" I changed the "set rssi_source = PROTOCOL instead of AUTO, and set rssi_max = 50, and all is working GREAT now. I hear INAV 2.4.0 corrected this issue.. 
THANKS to all JD

closed 2020-02-05 by DzikuVx