iNavFlight/inav #5312: PINIO enabled for more targets

pulquero, 2020-01-04:
## Current Behavior
Pinio seems to only be enabled on FCs that have a physical pin for it.

## Desired Behavior
Enable pinio on all F4s.

## Suggested Solution
Add USE_PINIO to common.h for F4s.

## Who does this impact? Who is this for?
Enables users to use resource mapping to remap existing pins for use by pinio.

## Additional context
A particular use case are vtxs with smartaudio/tramp pin and vtx power switcher pin. These two pins could be connected to a single uart, tx to sa/tramp and rx remapped to pinio and connected to power switcher.

See also betaflight.

MikeNH, 2020-10-01:
I bought a Runcam 2 4k thinking it is the perfect camera to install on an 8" iNav quad to be used as a photo/video platform. The setup instructions are pretty straightforward but, as it turns out, can't be installed on iNav so that video/photo can be controlled remotely. If this feature request was implemented, would this allow iNav to use this camera remote control as Betaflight can?

The Runcam 2 4k manual is here:
https://shop.runcam.com/runcam2-4k-edition/

Thank you

mric3412, 2020-12-07:
I have the same problem Mike (remote control with rc2-4K)
With fw v1.2.1 of runcam, I tested on the bench that a 100ms pulse from 5V to 0V starts/stop recording. Two pulses 5V>0V(100ms)>5V(100ms)>0V(100ms)>5v switches modes (video/photo). I generated these (inverted) pulses by using the circuit of an old 9g servo plugged on a PWM out of a standard receiver (100ms generated with opentx hrono function). Not tested for fw v1.2.2
If using a FC to control the camera, we have to use an appropriate output (5V switch signal)

closed 2022-05-07 by DzikuVx