iNavFlight/inav #5313: Navigation mode ATTI with big wind (copter)

jaguraphg, 2020-01-05:
Hi, please confirm if this is expected behavior or not. I configured the navigation modes, it seems to work fine. But today I was flying in the sufficient wind and realized the following.

1. When I fully released the sticks then the drone holds the position, altitude and the heading pretty fine.
2. When I change the altitude only by the throttle stick and do not touch roll/pitch then it also works pretty fine and the position is held properly.
3. If I try to move the copter towards or across the wind by the moving one of the stick (roll or pitch ) then it seems to try to move into the direction with the force in accordance of stick position but the position hold is released and the copter is being moved by the wind. So for example if I try to fly forward but the wind blows from the right side then the copter moves forward according to the level of the stick AND also moves to the left by the wind.

Could you please let me know if this is expected behavior or I missed something in the configuration.

Thanks.

jaguraphg, 2020-01-05:
Update:

INAV 2.3.0
Board: Kakute F7
GPS: SAM-M8Q GPS & QMC5883L

giacomo892, 2020-01-05:
You need to use control mode CRUISE to "fix" this.

closed 2020-01-05 by giacomo892

jaguraphg, 2020-01-05:
Just read about the CRUISE again and probably now I understand what this is for. I will check it tomorrow and will confirm if this is what I expected. Previously I was thinking that in CRUISE mode the speed or the copter is controlled by the angle and the angle is controlled by the stick like in ACRO mode so you release the stick but the copter continues the flight into the direction the stick was directed to. Thank you for your reply.