iNavFlight/inav #5315: Runway assist mode

Tygrys-1, 2020-01-05:
## Current Behavior
No such mode yet.

## Desired Behavior
For the wheeled aircraft/car/boat:
Hold the stright line (direction) using only the rudder or a stearing wheel. Like the "acro mode", but for the ground vehicle.

## Suggested Solution
Add the mode to the modes tab. The mode is enabled by the switch. After arming it will continue to keep the stright movement of the craft using a steering wheel or the rudder. The rudder channel input will make the correction to the direction, not do the actual steering. The up elevator input will switch the mode off, maybe after a delay. 

## Who does this impact? Who is this for?
For the wheeled aircraft/car/boat. To controll the aircraft in the straight line on the runway. It may also be used as the "acro mode" equivalent for cars or boats, when it may counteract the direction shift caused by the terrain or waves.

## Additional context
It should use the gyro to hold the direction. Can be a step to do the "ground missions".

monoxide13, 2020-01-15:
Have you tried using heading hold mode?

Tygrys-1, 2020-01-15:
Heading hold will not work. Inav uses roll to make turns, not the rudder. And the roll on the runway is not a good idea ;)

digitalentity, 2020-01-15:
A workaround is to use HEADING mode without TURN_ASSIST, but yeah, this could be an extension of takeoff mode

closed 2024-02-02 by DzikuVx